offensive-security/kali-arm-build-scripts #32: ./build-deps.sh E: everything

mturk2422, 2015-06-21:
heres what happens when i run the build-deps.sh (everything else up to this point went fine no problems, and i edited my /etc/apt/sources.list file to contain the proper sources for kali prior to beggining my set up) 

(before editing the sources.list)

root@localhost:/# cd lib/arm-linux-eabi
 root@localhost:/lib/arm-linux-eabi# git clone git://github.com/offensive-security/gcc-arm-eabi-linaro-4.6.2.git
 Cloning into 'gcc-arm-eabi-linaro-4.6.2'...
 remote: Counting objects: 333, done.
 remote: Total 333 (delta 0), reused 0 (delta 0), pack-reused 333
 Receiving objects: 100% (333/333), 37.56 MiB | 240 KiB/s, done.
 Resolving deltas: 100% (53/53), done.
 Checking out files: 100% (307/307), done.
 root@localhost:/lib/arm-linux-eabi# export PATH=${PATH}:/root/arm-stuff/gcc-arm-eabi-linaro-4.6.2/bin
 root@localhost:/lib/arm-linux-eabi# git clone git://github.com/offensive-security/kali-arm-build-scripts.git
 Cloning into 'kali-arm-build-scripts'...
 remote: Counting objects: 580, done.
 remote: Total 580 (delta 0), reused 0 (delta 0), pack-reused 580
 Receiving objects: 100% (580/580), 842.71 KiB | 243 KiB/s, done.
 Resolving deltas: 100% (391/391), done.
 root@localhost:/lib/arm-linux-eabi# ls
 gcc-arm-eabi-linaro-4.6.2  kali-arm-build-scripts
 root@localhost:/lib/arm-linux-eabi# cd kali-arm-build-scripts
 root@localhost:/lib/arm-linux-eabi/kali-arm-build-scripts# ./build-deps.sh
 Reading package lists... Done
 Building dependency tree
 Reading state information... Done
 Package gcc-multilib is not available, but is referred to by another package.
 This may mean that the package is missing, has been obsoleted, or
 is only available from another source

E: Package 'gcc-multilib' has no installation candidate
 E: Unable to locate package g++-multilib
 E: Couldn't find any package by regex 'g++-multilib'
 E: Unable to locate package pixz
 E: Unable to locate package cgpt
 E: Unable to locate package vboot-kernel-utils
 E: Unable to locate package vboot-utils
 E: Unable to locate package pixz
 Reading package lists... Done
 Building dependency tree
 Reading state information... Done
 libncurses5 is already the newest version.
 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 root@localhost:/lib/arm-linux-eabi/kali-arm-build-scripts#

(after editing the sources.list to contain the proper repos)

root@localhost:/lib/arm-linux-eabi/kali-arm-build-scripts# ./build-deps.sh                          

Reading package lists... Done
 Building dependency tree
 Reading state information... Done
 Package gcc-multilib is not available, but is referred to by another package.
 This may mean that the package is missing, has been obsoleted, or
 is only available from another source

E: Package 'gcc-multilib' has no installation candidate
 E: Unable to locate package g++-multilib
 E: Couldn't find any package by regex 'g++-multilib'
 Reading package lists... Done
 Building dependency tree
 Reading state information... Done
 libncurses5 is already the newest version.
 0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
 root@localhost:/lib/arm-linux-eabi/kali-arm-build-scripts#

Thoughts? Thank you in advance for the help, im fairly new to this but know enough to know theres an easy fix for this, i just dont know what it is.

steev, 2015-06-26:
It would help to know what type of machine you're doing this on.  Host OS, and so forth.  Details, details, details.  That said, I can only see this being an issue if perhaps the mirror you're hitting is out of date or there is an issue with it.  The scripts are meant to be run on an amd64 kali installation (though they will also run on x86, just won't compress the images), so if you're not using that, you MAY need to modify the package names to suit your distribution.

Also, putting everything in /lib is kind of frowned upon.  That's really not where things go.

DeveloppSoft, 2015-08-14:
Hoho...
When I ran it on my pi,  I got the same error, here is the fix:
Change gcc-multilib to gcc and g++-multilib to g++, if you have an other error, just remove g++ from the script.

steev, 2015-08-14:
Yeah, that's not going to work.  The script is meant to be run on amd64/x86 machines, not ARM devices.  Feel free to add in an if block for it and send a pull request though!

DeveloppSoft, 2015-08-14:
Walt. ..

DeveloppSoft, 2015-08-14:
Done, it might work

steev, 2015-08-14:
It *will* work, but seriously, you'd be better off running this on a machine that isn't an RPi.  Compiling the kernel on the RPi itself will take a very long time.  By all means go for it, but these scripts aren't designed for that purpose.

DeveloppSoft, 2015-08-14:
Yes ~ 12 hour

closed 2015-10-29 by steev

TheBlackRabbit, 2015-11-16:
Sounds to me like your host machine is running on ARM architecture... So, the master branch of the installer isn't going to work for you due to lack of multilibs, etc...

If a x64 host isn't an option for you, I would clone the newinstaller branch and run the python build. Dep-checks look for multilibs but won't find then however they were not necessary for me to successfully build. Any other packages "not found" may be in other ARM repo packages/meta-packages a quick apt-cache search or apt search should help them turn up.

I have been doing nightly nethunter builds via my Chromebook C100P. I also was able to run arm-build-scripts this way to build for veyron. Hope this helps you find out what some of your options are...

DeveloppSoft, 2015-11-17:
Some issues ago, I explained how to run on arm arch, please have a look.