offensive-security/kali-arm-build-scripts #33: Rpi B+ armhf

DeveloppSoft, 2015-08-09:
Hello guys,
I do not know if this is the right place to post that but I think it could be.
I think Kali Linux need an armhf image, si I have started to build one, based on Raspian, so I want to know if can pull the build script here.
Please note, I need the Kali armhf6 packages, so can build them and add to your repo ? or do I must to do it alone ?
Please, reply as soon as possible.

hcaijin, 2015-08-09:
Hello,
I think you may be wrong, you have to pack my side is to download from the
official website.

leto, 2015-08-09:
As far as I understand, the repo has a build script which generates what you want. Do you have a Raspberry Pi Model 2A ? I have one and I need to try Kali on it.

hcaijin, 2015-08-10:
My English is poor, I do not understand what you say the name of the first
half. I do not have Raspberry Pi Model 2A and may not help you. I just have
to build on my NEXUS4 over kali.

DeveloppSoft, 2015-08-10:
Sorry, but I am not wrong.
Currently kali support armel architecture for the rpi, but it is better to use the armhf architecture, wich is not currently supported, my goal is to bring to the community a solution to do that.
@hcaijin, me too, my english is not so good... Maybe I can speak french.
@leto, download the image from offensive security's site and burn it on to the sdcard, I think you might use the (2)B(+) model which is more powerful...
Actually, I have post that here because I do not know how to contact the kali linux developpers...

DeveloppSoft, 2015-08-10:
PS: So happy !, tomorrow kali 2.0 will be released, thanks to the developpers !!

leto, 2015-08-10:
Just so that everybody can understand this more:

arm and armhf have exactly the same features, but armhf has specialized hardware instructions which are virtualized in the arm instruction set. This means compilers on arm are quite inefficient at the expense of money, power use and human time. armhf is the better for power use and speed but arm will work across more things, i.e. has higher portability.

And yes, a huge :+1: :8ball: :+1: to all Kali devs! They are all welcome to crash on my couch in Portland, Oregon, USA.

steev, 2015-08-10:
(Official Kali ARM dev hat on)

Just to explain...

Currently there is *1* (2 if you count the EOL ODROID-W - and it can use the same exact software as the rpi so no point making a separate image) device that uses armv6 (armel in Debian) - it makes absolutely no sense for us to rebuild everything like the RPi foundation did for armhf support for the armv6.

Someone who wants to rebuild everything against the raspbian fork of Debian is more than welcome to, but with the sheer amount of devices we support (29 at the time of my writing this), doing all of that for *1* device (no matter how popular) just doesn't make sense.  

Sorry for any inconvenience, but I highly recommend at this point moving to either the ODROID-C or the RPi2 which are both armv7 and more performant than the RPi 1, for the same price.  I'm more than welcome to answer questions on IRC about things (#kali-linux my nick is steev) but this isn't going to officially happen from us.

closed 2015-08-10 by steev