offensive-security/kali-arm-build-scripts #34: Raspberry Pi 1 image doesn't allow headless ssh from scratch

KALRONG, 2015-08-12:
Hi,

Just tested the img for the raspberry pi, followed the simple steps of dd and then boot (headless) and everything was fine till I tried to connect over ssh, I just kept getting Access Denied.

The solution was to mount the SD on my laptop and modify the file /etc/ssh/sshd_config just changing this line:

PermitRootLogin without-password to PermitRootLogin yes

After that it worked just fine.

Hope this helps if somebody else has this problem.

Best regards.

steev, 2015-08-12:
Correct, although the more proper fix is to copy your id_{r,d}sa.pub into /root/.ssh/authorized_keys on the root partition.  This was a change that was made since many people leave the password the default "toor" so this way, people can still be lazy about not changing the password without people being able to ssh in.

steev, 2015-08-12:
After discussing with muts, we've decided to go ahead and make it easier for people and re-enabling ssh login as root with a password by default.  This change is commit 7ffffb9 - I prefer the more proper fix of using the authorized_keys, but apparently that's too hard for people.

closed 2015-08-12 by steev