offensive-security/kali-arm-build-scripts #36: Arm dev

DeveloppSoft, 2015-08-14:
Hi, just to say that I am starting developping the armhf6 version of kali based on raspbian jessie.
I just need an easy way to download pkg sources.

steev, 2015-08-15:
I don't typically deal with that side of things, but you'll want to read 

https://www.debian.org/doc/manuals/maint-guide/index.en.html

That being said, this isn't really a forum for support for that.  Please use the Kali forums, or (more appropriately, you'll want to create your own)  as your fork will *not* be Kali.

closed 2015-08-15 by steev

steev, 2015-08-15:
Since you attempted to PM me on IRC while I was asleep I'll reply here - what you are doing is NOT Kali - it is taking the raspbian base and applying (possibly) the kali patches on top.  It cannot be called Kali as we have no part in this work.  Please see https://www.kali.org/trademark-policy/ to understand the Kali trademark better.