offensive-security/kali-arm-build-scripts #38: Blackscreen

phr34k0, 2015-09-05:
Booting off Kali 2.0 from my Raspi B and a TFT touchscreen gives a black blank screen. Desktop only appears in HDMI mode. Changing TTY mode with ALT+CTRL+F1(to F7) becomes responsive only once logged in on desktop

phr34k0, 2015-09-05:
PITFT touchscreen btw

steev, 2015-09-05:
The PiTFT image requires some setup since there are multiple TFT screens - you'd need to look up what you need to change to get it to show up; you likely just need to export the display properly or set the cmdline differently.

phr34k0, 2015-09-06:
Fixed it with 

root@kali:~# cp /etc/X11/default-display-manager /etc/X11/default-display-manager.bak
root@kali:~# echo > /etc/X11/default-display-manager

as i'm using the kali-image with TFT support from kali website. Appears that the above was the reason why it 'interferes' and automatically go to desktop mode using HDMI option instead of TFT screen. By booting it to Console Mode and manually startx fixes the problem.

closed 2015-10-29 by steev