offensive-security/kali-arm-build-scripts #42: BeagleBone Black: g_ether module not loaded by default

strasharo, 2015-11-06:
I've downloaded the Kali image from here http://images.kali.org/kali-2.0.1-bbb.img.xz (version 2.0.1 checksum c1b6b98eca4fa75fc24caff98adea27d5a82c553) and by default the module g_ether is not being loaded on my BBB. When I modprobe it manually everything works. I see that there's already a script included for this, but there's a comment that it's currently unused. I think that it would be useful if we have that module enabled by default as all of the other distros for BBB do. Snippet below:

# Unused currently, but this script is a part of using the usb as an ethernet
# device.
wget -c https://raw.github.com/RobertCNelson/tools/master/scripts/beaglebone-black-g-ether-load.sh -O ${basedir}/root/root/beaglebone-black-g-ether-load.sh
chmod +x ${basedir}/root/root/beaglebone-black-g-ether-load.sh

steev, 2015-11-09:
I'm not entirely sure what you're trying to say here.  Can you please clarify?

strasharo, 2015-11-11:
In the default installation of Kali 2.0.1 on BeagleBone Black we have no configuration to load the g_ether module by default. It would be helpful for the initial setup of the device, if we load the module by default as it's being done in the default Ångström distribution for example.

duchengyao, 2016-02-11:
`apt-get install udhcpd ` before running `beaglebone-black-g-ether-load.sh`

steev, 2016-02-12:
@duchengyao what do you mean?  if you're trying to do it locally with your script, just add udhcpd to the extras line at the top of the script for packages to install.  

The images aren't meant to be the end all be all that we release, and the scripts are here for users to customize to their hearts content.

duchengyao, 2016-02-12:
@steev I mean, `~/beaglebone-black-g-ether-load.sh` doesn't work in the original `kali-bbb-XXX.img`, because of missing  udhcpd.  And no error message output.

steev, 2016-02-12:
Install udhcpd? It's not my g_ether script, so I don't know why it doesn't give output.  I've added udhcpd to the build script so future images will have it pre-installed.  I'm not sure what exactly more I can do here.

duchengyao, 2016-02-12:
"Added udhcpd to the build script" is a good idea. Thanks so much ^_^

steev, 2016-02-12:
Sorry, I had pushed to my repo and forgot to send the pull request - it's merged here now.  Closing.

closed 2016-02-12 by steev