offensive-security/kali-arm-build-scripts #43: BeagleBone Black: Unable to load g_serial module

strasharo, 2015-11-06:
I'm using the latest BeagleBone Black image: (http://images.kali.org/kali-2.0.1-bbb.img.xz), however I'm unable insert the g_serial module for some reason:

root@kali:~# modprobe g_serial
modprobe: ERROR: could not insert 'g_serial': No such device
root@kali:~# modinfo g_serial
filename:       /lib/modules/3.8.13-bone53/kernel/drivers/usb/gadget/g_serial.ko
license:        GPL
author:         David Brownell
author:         Al Borchers
description:    Gadget Serial v2.4
license:        GPL
author:         Felipe Balbi
srcversion:     A7058FBDCF3A5E10C147597
depends:        libcomposite
intree:         Y
vermagic:       3.8.13-bone53 SMP mod_unload modversions ARMv7 thumb2 p2v8
parm:           idVendor:USB Vendor ID (ushort)
parm:           idProduct:USB Product ID (ushort)
parm:           bcdDevice:USB Device version (BCD) (ushort)
parm:           iSerialNumber:SerialNumber string (charp)
parm:           iManufacturer:USB Manufacturer string (charp)
parm:           iProduct:USB Product string (charp)
parm:           use_acm:Use CDC ACM, default=yes (bool)
parm:           use_obex:Use CDC OBEX, default=no (bool)
parm:           n_ports:number of ports to create, default=1 (uint)
root@kali:~#

steev, 2015-11-09:
Does it work for you if you use "modprobe g_serial stall=0 removable=1 nofua=1"  I haven't tried the g_serial driver myself as I normally just use an actual adapter connected to the pins, but I'm looking into this.

strasharo, 2015-11-11:
root@kali:~# modprobe g_serial stall=0 removable=1 nofua=1
modprobe: ERROR: could not insert 'g_serial': Invalid argument
root@kali:~#

steev, 2016-02-12:
Does this still fail to work for you with the 2.1 release?

strasharo, 2016-03-01:
The module is no longer present in 2.1:

root@kali:~# modprobe g_serial
modprobe: FATAL: Module g_serial not found in directory /lib/modules/4.4.0-armv7-devel-r28
root@kali:~# lsb_release -a
No LSB modules are available.
Distributor ID:	Kali
Description:	Kali GNU/Linux Rolling
Release:	kali-rolling
Codename:	kali-rolling
root@kali:~# uname -a;date;uptime
Linux kali 4.4.0-armv7-devel-r28 #2 SMP Sat Jan 23 02:53:51 CST 2016 armv7l GNU/Linux
Tue Mar  1 12:10:40 UTC 2016
 12:10:40 up 7 min,  1 user,  load average: 1.07, 0.75, 0.36
root@kali:~#

steev, 2016-03-01:
There is a 2.1.1 image that hasn't been posted yet, unfortunately, and the g_serial module does exist there (just checked).  Please try again once it's posted.

strasharo, 2016-03-01:
Sure, will give it a go once the image is available and I'll report here the status.
Thank you.

strasharo, 2016-03-26:
```
root@kali:~# uname -a;date;uptime
Linux kali 4.1.18 #1 SMP PREEMPT RT Wed Mar 16 20:59:54 CDT 2016 armv7l GNU/Linux
Sat Mar 26 16:50:17 UTC 2016
 16:50:17 up 4 min,  1 user,  load average: 0.07, 0.11, 0.05
root@kali:~# modinfo g_serial
filename:       /lib/modules/4.1.18/kernel/drivers/usb/gadget/legacy/g_serial.ko
license:        GPL
author:         David Brownell
author:         Al Borchers
description:    Gadget Serial v2.4
depends:        libcomposite
intree:         Y
vermagic:       4.1.18 SMP preempt mod_unload modversions ARMv7 thumb2 p2v8 
parm:           idVendor:USB Vendor ID (ushort)
parm:           idProduct:USB Product ID (ushort)
parm:           bcdDevice:USB Device version (BCD) (ushort)
parm:           iSerialNumber:SerialNumber string (charp)
parm:           iManufacturer:USB Manufacturer string (charp)
parm:           iProduct:USB Product string (charp)
parm:           use_acm:Use CDC ACM, default=yes (bool)
parm:           use_obex:Use CDC OBEX, default=no (bool)
parm:           n_ports:number of ports to create, default=1 (uint)
root@kali:~# modprobe g_serial
modprobe: ERROR: could not insert 'g_serial': No such device
root@kali:~# modprobe g_serial stall=0 removable=1 nofua=1
modprobe: ERROR: could not insert 'g_serial': No such device
root@kali:~# 

```