offensive-security/kali-arm-build-scripts #45: bbb.sh - kernel source doesnt exist anymore

0xicl33n, 2016-01-23:
Line 202 - `git clone --depth 1 --branch am33x-v3.8 https://github.com/RobertCNelson/linux-dev ${basedir}/kernel`

Branch am33x-v3.8 does not exist anymore

https://github.com/RobertCNelson/stable-kernel/tree/v3.8.x 

This seems to be the new source. I will try building an image with it now

steev, 2016-01-26:
They will be - I was in the process of building and testing the release based on kali-rolling.  Appreciate the report though!  The new one is based on the master branch of linux-dev which uses a 4.4 kernel.

0xicl33n, 2016-01-28:
Sounds good & cant wait! Could you possibly do something about ssh as root being disabled on boot because some of us dont have microhdmi cables to connect the board up to a monitor and get in there manually

steev, 2016-01-28:
Yep, that's one of the fixes coming with 2.1 as well :)

duchengyao, 2016-02-01:
New Kernel，like it !

closed 2016-02-01 by steev