offensive-security/kali-arm-build-scripts #47: bbb，mouse & HDMI  crash

duchengyao, 2016-02-01:
1. In new bbb 2.1, the bug [!](https://bugs.kali.org/view.php?id=2510 are still exist)
2. No HDMI output. I am using microHDMI to VGA adapter to my old screen, the screen could detect the signal but nothing display.

             Starting udev Coldplug all Devices...
             Starting Flush Journal to Persistent Storage...
        [  OK  ] Reached target Local File Systems (Pre).
        [  OK  ] Started Load/Save Random Seed.
        [    9.782764] systemd-journald[124]: Received request to flush runtime journal from PID 1
        [  OK  ] Started Flush Journal to Persistent Storage.
        [  OK  ] Started udev Coldplug all Devices.
        [  OK  ] Found device /dev/ttyO0.
        [   11.403181] tda998x 0-0070: Falling back to first CRTC
        [   11.409405] cpu cpu0: of_pm_voltdm_notifier_register: Failed to get cpu0 regulator/voltdm: -517
        [   11.430277] cpu cpu0: cpu0 clock notifier not ready, retry
        [   11.576098] tda998x 0-0070: found TDA19988
        [   11.610799] tilcdc 4830e000.lcdc: bound 0-0070 (ops tda998x_ops [tda998x])
        [   11.618040] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).
        [   11.625096] [drm] No driver support for vblank timestamp query.

steev, 2016-02-02:
Can you provide the full dmesg output?  And Xorg log file please?

0xicl33n, 2016-02-02:
I have a feeling this is because kernel patches arent working with bbb.sh, i was going to make my own issue but:

--------------------------
File to patch:
Skip this patch? [y] y
Skipping patch.
1 out of 1 hunk ignored
+ Detected build host [Kali GNU/Linux Rolling]
+ host: [x86_64]
+ git HEAD commit: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
Dependency check skipped, you are on your own.
-----------------------------
Unrecognized deb based system:
-----------------------------
Please cut, paste and email to: bugs@rcn-ee.com
-----------------------------
git: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
git: [url=https://github.com/RobertCNelson/linux-dev]
uname -m: [x86_64]
lsb_release -a:
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        kali-rolling
Codename:       kali-rolling
-----------------------------
* Failed dependency check
cp: cannot stat '/build/kali-arm-build-scripts/beaglebone-black-1.2/kernel/deploy/4.*.zImage': No such file or directory
tar: /build/kali-arm-build-scripts/beaglebone-black-1.2/kernel/deploy/4.*-dtbs.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
tar: /build/kali-arm-build-scripts/beaglebone-black-1.2/kernel/deploy/4.*-modules.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
Cloning into 'firmware'...
fatal: '/root/sandbox/mirror/linux-firmware.git' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.
tar: /build/kali-arm-build-scripts/beaglebone-black-1.2/kernel/deploy/4.*-firmware.tar.gz: Cannot open: No such file or directory
tar: Error is not recoverable: exiting now
--2016-02-01 18:02:34--  https://raw.github.com/RobertCNelson/tools/master/scripts/beaglebone-black-g-ether-load.sh
Resolving raw.github.com (raw.github.com)... 23.235.47.133
Connecting to raw.github.com (raw.github.com)|23.235.47.133|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://raw.githubusercontent.com/RobertCNelson/tools/master/scripts/beaglebone-black-g-ether-load.sh [following]
--2016-02-01 18:02:35--  https://raw.githubusercontent.com/RobertCNelson/tools/master/scripts/beaglebone-black-g-ether-load.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 199.27.79.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|199.27.79.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 3811 (3.7K) [text/plain]
Saving to: '/build/kali-arm-build-scripts/beaglebone-black-1.2/root/root/beaglebone-black-g-ether-load.sh'

/build/kali-arm-build-scripts/beaglebone-bla 100%[===========================================================================================>]   3.72K  --.-KB/s    in 0s

2016-02-01 18:02:35 (113 MB/s) - '/build/kali-arm-build-scripts/beaglebone-black-1.2/root/root/beaglebone-black-g-ether-load.sh' saved [3811/3811]

del devmap : loop0p2
del devmap : loop0p1
Removing temporary build files
Generating sha1sum for kali-1.2-bbb.img
Compressing kali-1.2-bbb.img

steev, 2016-02-02:
Double check that it cloned the kernel correctly - I had that happen to me at one point where my internet hiccuped and it didn't clone the linux kernel during that step.

steev, 2016-02-02:
Ah, I just noticed you're not the original filer - yes, your issue is unrelated.  Please do not file unrelated items under unrelated bugs.  That should be it's own issue, but as I noted - if it doesn't clone the kernel (the linux-dev repo does its own clone of a kernel) properly, then it can't patch anything.

0xicl33n, 2016-02-02:
Sorry - removing post and i shall make an issue if this build doesnt work

duchengyao, 2016-02-02:
Welcome to minicom 2.7

	OPTIONS: I18n 
	Compiled on Nov 30 2015, 11:52:37.
	Port /dev/ttyUSB0, 11:02:50

	Press CTRL-A Z for help on special keys

	U-Boot SPL 2015.10-00001-g51aa16b (Oct 21 2015 - 14:16:21)
	bad magic

	U-Boot 2015.10-00001-g51aa16b (Oct 21 2015 - 14:16:21 -0500), Build: jenkins-gi0

	       Watchdog enabled
	I2C:   ready
	DRAM:  512 MiB
	Reset Source: Power-on reset has occurred.
	MMC:   OMAP SD/MMC: 0, OMAP SD/MMC: 1
	Using default environment

	Net:   <ethaddr> not set. Validating first E-fuse MAC                           
	cpsw                                                                            
	Hit any key to stop autoboot:  0                                                
	gpio: pin 53 (gpio 53) value is 1                                               
	switch to partitions #0, OK                                                     
	mmc0 is current device                                                          
	gpio: pin 54 (gpio 54) value is 1                                               
	Checking for: /uEnv.txt ...                                                     
	reading uEnv.txt                                                                
	1116 bytes read in 6 ms (181.6 KiB/s)                                           
	gpio: pin 55 (gpio 55) value is 1                                               
	Loaded environment from uEnv.txt                                                
	Importing environment from mmc ...                                              
	Checking if uenvcmd is set ...                                                  
	gpio: pin 56 (gpio 56) value is 1                                               
	Running uenvcmd ...                                                             
	reading zImage                                                                  
	5865816 bytes read in 323 ms (17.3 MiB/s)                                       
	reading /dtbs/am335x-boneblack.dtb                                              
	57149 bytes read in 12 ms (4.5 MiB/s)                                           
	Kernel image @ 0x82000000 [ 0x000000 - 0x598158 ]                               
	## Flattened Device Tree blob at 88000000                                       
	   Booting using the fdt blob at 0x88000000                                     
	   Loading Device Tree to 8ffef000, end 8fffff3c ... OK                         
	                                                                                
	Starting kernel ...                                                             
	                                                                                
	[    0.000000] Booting Linux on physical CPU 0x0                                
	[    0.000000] Linux version 4.4.0-armv7-devel-r28 (root@builder) (gcc version 6
	[    0.000000] CPU: ARMv7 Processor [413fc082] revision 2 (ARMv7), cr=10c5387d  
	[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instructie
	[    0.000000] Machine model: TI AM335x BeagleBone Black                        
	[    0.000000] cma: Reserved 24 MiB at 0x9e000000                               
	[    0.000000] Memory policy: Data cache writeback                              
	[    0.000000] CPU: All CPU(s) started in SVC mode.                             
	[    0.000000] AM335X ES2.0 (sgx neon )                                         
	[    0.000000] PERCPU: Embedded 13 pages/cpu @df926000 s23552 r8192 d21504 u5328
	[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pa8
	[    0.000000] Kernel command line: console=ttyO0,115200n8 root=/dev/mmcblk0p2 c
	[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)              
	[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)  
	[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)   
	[    0.000000] Memory: 472068K/522240K available (12312K kernel code, 1294K rwd)
	[    0.000000] Virtual kernel memory layout:                                    
	[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)                
	[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)                
	[    0.000000]     vmalloc : 0xe0800000 - 0xff800000   ( 496 MB)                
	[    0.000000]     lowmem  : 0xc0000000 - 0xe0000000   ( 512 MB)                
	[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)                
	[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)                
[... 344 lines cut ...]
	         Starting udev Kernel Device Manager...                                 
	[  OK  ] Started udev Kernel Device Manager.                                    
	         Starting LSB: Set preliminary keymap...                                
	[  OK  ] Started LSB: Set preliminary keymap.                                   
	         Starting Remount Root and Kernel File Systems...                       
	[    9.242805] EXT4-fs (mmcblk0p2): re-mounted. Opts: errors=remount-ro         
	[  OK  ] Started Remount Root and Kernel File Systems.                          
	         Starting udev Coldplug all Devices...                                  
	         Starting Load/Save Random Seed...                                      
	[  OK  ] Reached target Local File Systems (Pre).                               
	         Starting Flush Journal to Persistent Storage...                        
[... cut ...]
	[   11.623651] tda998x 0-0070: found TDA19988                                   
	[   11.651208] tilcdc 4830e000.lcdc: bound 0-0070 (ops tda998x_ops [tda998x])   
	[   11.658446] [drm] Supports vblank timestamp caching Rev 2 (21.10.2013).      
	[   11.665511] [drm] No driver support for vblank timestamp query.              

Notice: this crash is from the img downloaded from [kali-2.1-bbb.img.xz](https://images.offensive-security.com/arm-images/kali-2.1-bbb.img.xz). And I have tried *am335x-boneblack-hdmi-overlay.dtb*, it also has the same issue

duchengyao, 2016-02-02:
And could you pleas tell me how to find the Xorg log?
I removed Xorg.0.log & Xorg.0.log.old from /var/log , restart the BBB, but I can't find new Xorg.log file.

steev, 2016-02-02:
That should happen automatically when X starts on the box - if not, perhaps logging in first?

Can you tell me how you are powering your bbb, and what devices you are plugging in, and what type of monitor you're using?  

And if you can log in on the serial console, can you provide the actual dmesg output?  Reading through the systemd spam is rather annoying (and not everything printed to the screen is what shows up in dmesg.  Thanks!

duchengyao, 2016-02-02:
BBB, 5V power plugin, and microHDMI to VGA adapter, VGA monitor

If not plugging micro HDMI ,  I can log in the serial console; but if  i plug my monitor,  the BBB stoped when the serial output *[   11.665511] [drm] No driver support for vblank timestamp query. * ,and I can't log in on the serial console.

I am trying to get the actual dmesg output...

steev, 2016-02-02:
Have you seen https://www.youtube.com/watch?v=toPYh_VCHvI ?

duchengyao, 2016-02-02:
I used the BBB direct a hdmi monitor, but still nothing display.
Have you ever tested the img? 
Or, I'm using the wrong dtb file?

steev, 2016-02-02:
Yes, I test every image before release and it worked here with my Samsung monitor via microHDMI->HDMI connection.

duchengyao, 2016-02-02:
Which dtb I should choose? the original am335x-boneblack.dtb?
Thanks!

steev, 2016-02-02:
Correct, you should only be using the cape one if you're *actually* using a cape.  The correct one should be picked up by u-boot, but if you're randomly renaming files trying to get it to work, it's not going to work.

duchengyao, 2016-02-03:
I have read the log, and found a ERROR.  

        [    4.485230] [drm] Initialized drm 1.1.0 20060810                             
        [    4.490976] Error: Driver 'tfp410' is already registered, aborting...        

The 'tfp410' may be a DVI interface [FP410 TI PanelBus™ Digital Transmitter](http://www.ti.com/lit/ds/symlink/tfp410.pdf)

When I am using the old kali img（kernel 3.8.13) , the message is that:

        [    2.340469] [drm] Supports vblank timestamp caching Rev 1 (10.10.2010).
        [    2.347392] [drm] No driver support for vblank timestamp query.
        [    2.447805] tilcdc 4830e000.fb: Not sure which CEA mode to set, leaving as DVI

duchengyao, 2016-02-12:
@steev Notice! This issue is not related to [Add an xorg.conf to force 16bit video and fbdev](https://github.com/steev/kali-arm-build-scripts/commit/3d7fd9be36281bd9ef3cee3e59eb3a76e67d1670) !

It is a kernel problem. I  replaced `am335x-boneblack.dtb` and `zImage` from `kali-bbb-1.1.0.img (kernel 3.8.13)`; hdmi started to work!!!

steev, 2016-02-13:
So you're saying with the 4.4 kernel, it fails to do any HDMI for you?

duchengyao, 2016-02-13:
Yes. And even if hdmi is not plugging in, the log is :

        [    4.485230] [drm] Initialized drm 1.1.0 20060810                             
        [    4.490976] Error: Driver 'tfp410' is already registered, aborting...   

And I found `cpufreq-info` fail, in the 4.4 kernel. I am doing another test to confirm this issue

elationfoundation, 2016-02-16:
I also am unable to use HDMI on my beagle bone black. kernel version `4.4.0-armv7-devel-r28`.

duchengyao, 2016-02-20:
I've noticed that the latest image in beagleboard.org is using kernel 4.1.17. I think Kernel 4.1.X is a better choice for Beaglebone Black. 
http://elinux.org/Beagleboard:BeagleBoneBlack_Debian

steev, 2016-02-20:
I'm currently running a build against the 4.1 kernel sources - once I finish it, if you'd like to give it a try, shoot me an email. steev at kali dot org

duchengyao, 2016-02-23:
I tried your new image `kali-2.1.1-bbb.img.xz` with kernel 4.1.17; HDMI works !

hagensieker, 2016-02-24:
2.1.1 also please.  I just bought a BBB and downloaded 2.1 and no HDMI also.  Thanks

hackerz217, 2016-02-24:
hey,i know this question has nothing to do with this post but can u tell me how can i install airdrop-ng in kali rolling 2016,i tried all ways google said and m receiving all kinda errors,help plz

steev, 2017-10-02:
Can one of you that was having hdmi issues please test the 2017.2 issue and let me know if you are still having issues?

closed 2017-10-02 by steev