offensive-security/kali-arm-build-scripts #48: bbb.sh - weird kernel problems (it does not build)

0xicl33n, 2016-02-02:
```
Cloning into '/build/kali-arm-build-scripts/beaglebone-black-1.3/kernel'...
remote: Counting objects: 223, done.
remote: Compressing objects: 100% (208/208), done.
remote: Total 223 (delta 32), reused 80 (delta 11), pack-reused 0
Receiving objects: 100% (223/223), 667.61 KiB | 0 bytes/s, done.
Resolving deltas: 100% (32/32), done.
Checking connectivity... done.
+ Detected build host [Kali GNU/Linux Rolling]
+ host: [x86_64]
+ git HEAD commit: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
Dependency check skipped, you are on your own.
-----------------------------
Unrecognized deb based system:
-----------------------------
Please cut, paste and email to: bugs@rcn-ee.com
-----------------------------
git: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
git: [url=https://github.com/RobertCNelson/linux-dev]
uname -m: [x86_64]
lsb_release -a:
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        kali-rolling
Codename:       kali-rolling
-----------------------------
* Failed dependency check
./bbb.sh: line 210: cd: /build/kali-arm-build-scripts/beaglebone-black-1.3/kernel/KERNEL: No such file or directory
can't find file to patch at input line 5
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|diff --git a/net/mac80211/tx.c b/net/mac80211/tx.c
|index 0855cac..221bed6 100644
|--- a/net/mac80211/tx.c
|+++ b/net/mac80211/tx.c
--------------------------
File to patch:
```

```
root@lead:/build/kali-arm-build-scripts/beaglebone-black-1.3/kernel# ./build_kernel.sh
+ Detected build host [Kali GNU/Linux Rolling]
+ host: [x86_64]
+ git HEAD commit: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
Dependency check skipped, you are on your own.
-----------------------------
Unrecognized deb based system:
-----------------------------
Please cut, paste and email to: bugs@rcn-ee.com
-----------------------------
git: [9b9c936a2950210e70d9d1cd37e60088c0dff314]
git: [url=https://github.com/RobertCNelson/linux-dev]
uname -m: [x86_64]
lsb_release -a:
No LSB modules are available.
Distributor ID: Kali
Description:    Kali GNU/Linux Rolling
Release:        kali-rolling
Codename:       kali-rolling
-----------------------------
* Failed dependency check

root@lead:~# cat /etc/issue
Kali GNU/Linux Rolling \n \l
root@lead:~# uname -a
Linux lead 4.0.0-kali1-amd64 #1 SMP Debian 4.0.4-1+kali2 (2015-06-03) x86_64 GNU/Linux

```

steev, 2016-02-02:
Can you paste (or pastebin, or gist) the full output of the build script please?  

I doubt it's related to rolling, but I should note that I currently build everything on a sana based buildbox.

0xicl33n, 2016-02-02:
Ive been building  it in a screen session and using putty from this laptop  but i can probably just `>>` to a file...right?

steev, 2016-02-02:
Actually - it may entirely be kali-rolling related - the build script from robert might not yet support kali-rolling as a supported distro.  Realistically, we should move to doing something similar to what http://ragustechblog.blogspot.com/2014/12/how-to-compile-latest-mainline-kernel.html mentions (creating the uImage this way)

If you're building in screen, before you run a build just do ctrl+a, :log and it should log everything to a file "screenlog" in $CWD that screen is run from.

steev, 2016-02-02:
Actually - we don't even need to be doing a uImage, the zImage and friends should work just fine since that's how it is currently done.  It's just a matter of putting them all in the correct places.

0xicl33n, 2016-02-02:
I actually just shut down my build server from frustration so now ill have to run another build but thats no problem here, it may take about 30-45 minutes to get to that point..

I used to have apt-cacher setup but its gone now

Rebooting the build server

steev, 2016-02-02:
Don't bother - notice the message that I skipped over - unknown deb based system, copy/paste the following output and send to bugs@ - that means that's why it is failing to build.  It builds for me because I'm still on sana.

0xicl33n, 2016-02-02:
Actually 

I found  in `linux-dev/tools/host_det.sh` the following:

```
if [ "x${deb_distro}" = "xsana" ] ; then
			#lsb_release -a
			#Distributor ID:    Kali
			#Description:    Kali GNU/Linux 2.0
			#Release:    2.0
			#Codename:    sana
			deb_distro="jessie"
		fi
```

After a little poking around the config for rolling should look like:

```

if [ "x${deb_distro}" = "kali-rolling" ] ; then
			#lsb_release -a
			#Distributor ID: Kali
			#Description:    Kali GNU/Linux Rolling
			#Release:        kali-rolling
			#Codename:       kali-rolling

			deb_distro="sid"
		fi
```
( im assuming rolling is in fact sid)

steev, 2016-02-02:
Yes, but that's his repo not ours.... so, you can make the modification locally and copy the file in as part of the build script, or wait for his update to the file.

0xicl33n, 2016-02-02:
Indeed, i will test with a local copy and perhaps make a pull request

steev, 2016-02-02:
He happily accepts them :)

closed 2016-02-02 by steev