offensive-security/kali-arm-build-scripts #49: bbb: lack xorg.conf ?

duchengyao, 2016-02-03:
I have a Kali 2.0 & 2.1 problem [0002510: Beaglebone mouse crash](https://bugs.kali.org/view.php?id=2510) for several month.
And I created xorg.conf following this comment [41#issuecomment-178980188](https://github.com/RobertCNelson/linux-dev/issues/41#issuecomment-178980188) to solve my problem perfectly.

steev, 2016-02-03:
Okay, I'm working on a new image incorporating forcing fbdev on.  Thanks for pointing that out, and to Robert for the way to get it to happen here.  I wasn't seeing the issue so it was hard to reproduce but could when he mentioned opening a jpeg in a gtk2 app.

duchengyao, 2016-02-03:
By using this method, although my BBB is never crash, the resolution and colour is not correct, when using a BBB-EXP 7' LCD. Please notice that.

Thanks so much.

steev, 2016-02-03:
You'll probably need a more specific xorg setup to set the correct resolution and display color depth - the one you linked sets it to 16bit color, so you may need more, or less - however, it may also just be a kernel bug that it's setting incorrect colors, in which case, you need to contact the mainline kernel developers and get a fix in the kernel for that.

duchengyao, 2016-02-03:
Thanks for your explanation. 
I still have a question. We are using the same device, the same image, why just I and [Fredo](https://bugs.kali.org/view.php?id=2510) have the Xorg crash; nobody else have the same issue?

RobertCNelson, 2016-02-03:
onboard hdmi, lcd4/lcd7 need 16..
bbb-exp, bbb-view need 24..

steev, 2016-02-03:
I can't say why it happens for you and Fredo.  I'm able to trigger it too using the information that Robert posted on the other bug. So I don't think it's so much a bad device as you're hitting the race and I'm somehow not. 

And regarding his comment here, you may be able to create an xorg.conf.d directory in /etc/X11/ and put a .conf snippet in there to set the depth to 24 when using the -exp or -view.

closed 2016-02-11 by duchengyao