offensive-security/kali-arm-build-scripts #50: bbb.sh - no firmware.git?

0xicl33n, 2016-02-03:
I just cannot get these to build correctly, im even using sana now in 64 bit. 
 
```
Cloning into 'firmware'...
fatal: '/root/sandbox/mirror/linux-firmware.git' does not appear to be a git repository
fatal: Could not read from remote repository.

Please make sure you have the correct access rights
and the repository exists.

..rest of "build"
```

I have a full build log https://gist.githubusercontent.com/0xicl33n/f9a0782da63058b3d1e1/raw/ccfad6fe01a87cbcdd2bf6ca8632218e08878528/screenlog.0

duchengyao, 2016-02-04:
And the bbb.sh line 205 `export LINUX_GIT=/root/sandbox/mirror/mainline.git` may be something wrong.

0xicl33n, 2016-02-04:
Ive been staring at that line for a very long and even tried googling it to no avail. i have no idea what that line is about because  it isnt referenced again in the code

duchengyao, 2016-02-04:
I have noticed ${basedir}/kernel/Kernel ( linux-dev.git ) 

README:

        By default this script will clone the linux-stable tree:
        https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git
        to: ${DIR}/ignore/linux-src:
        
        If you've already cloned torvalds tree and would like to save some hard drive
        space, just modify the LINUX_GIT variable in system.sh to point to your current
        git clone directory.

system.sh.sample Line 22-24

        ###OPTIONAL: LINUX_GIT: specify location of locally cloned git tree.
        #
        #LINUX_GIT=/home/user/linux-stable/

0xicl33n, 2016-02-04:
OH MY GOD I FEEL DUMB

duchengyao, 2016-02-04:
In other script like chromebook-arm-acer.sh, rpi.sh, the line is `git clone https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git firmware`

steev, 2016-02-10:
Oops, since I build the images on a box, they all point to local checkouts when I build the image so that I can guarantee the same versions of packages and gits across the boards.  Sorry about that!

closed 2016-02-10 by steev

duchengyao, 2016-02-11:
@steev Thanks a lot!