offensive-security/kali-arm-build-scripts #52: utilite.sh - utilite-1.0/root/etc/init.d/zram': No such file or directory

AaronNGray, 2016-02-07:
Building Utilite 1.0 on Kali-2016 an error has creeped in with /etc/init.d/zram missing
```
Checking connectivity... done.
cp: cannot create regular file '/root/Utilite/kali-arm-build-scripts/utilite-1.0/root/etc/init.d/zram': No such file or directory
chmod: cannot access '/root/Utilite/kali-arm-build-scripts/utilite-1.0/root/etc/init.d/zram': No such file or directory
del devmap : loop0p2
device-mapper: remove ioctl on loop0p1 failed: Device or resource busy
```

steev, 2016-02-10:
I can't seem to reproduce this here.  Does your path to the image you're building include a space?  The scripts aren't particularly resilient and don't expect a space in the path name.

AaronNGray, 2016-02-10:
okay not at base at mo will check when I return. I will rerun scripts in
fresh directory checking all details too.

Many Thanks !

steev, 2016-02-10:
Cloning into 'firmware'...
remote: Counting objects: 1256, done.
remote: Compressing objects: 100% (929/929), done.
remote: Total 1256 (delta 314), reused 1214 (delta 293)
Receiving objects: 100% (1256/1256), 58.99 MiB | 12.67 MiB/s, done.
Resolving deltas: 100% (314/314), done.
Checking connectivity... done.
del devmap : loop0p2
del devmap : loop0p1

One thing you may want to do is scroll back through the output - did it even create the image correctly for you?  I'm saying that based on the device-mapper having an issue with deleting loop0p1.  Make sure you've run the build-deps.sh script so that it pulls in the tools that are needed.

AaronNGray, 2016-02-11:
From what I can remember there were no other errors, will double check
though. I have not run the image yet though.

AaronNGray, 2016-02-11:
Okay now I am getting another error. I am running 32bit Kali-2016 in VMWare on a 64Bit Windows 10 Machine

```
root@kali:~/build/kali-arm-build-scripts# git clone https://github.com/offensive-security/kali-arm-build-scripts.git
root@kali:~/build/kali-arm-build-scripts# git clone https://github.com/offensive-security/gcc-arm-linux-gnueabihf-4.7.git
root@kali:~/build/kali-arm-build-scripts# cd kali-arm-build-scripts/
root@kali:~/build/kali-arm-build-scripts# PATH=/root/build/gcc-arm-linux-gnueabihf-4.7/bin:${PATH}
root@kali:~/build/kali-arm-build-scripts# ./build-deps.sh
root@kali:~/build/kali-arm-build-scripts# /utilite.sh 1.0 2>&1 | tee out.txt
```
[out.txt](https://github.com/offensive-security/kali-arm-build-scripts/files/126984/out.txt)

```
root@kali:~/build/kali-arm-build-scripts# df
Filesystem          1K-blocks     Used Available Use% Mounted on
udev                    10240        0     10240   0% /dev
tmpfs                  827736     9572    818164   2% /run
/dev/sda1            47353196 14156208  30768524  32% /
tmpfs                 2069336      496   2068840   1% /dev/shm
tmpfs                    5120        0      5120   0% /run/lock
tmpfs                 2069336        0   2069336   0% /sys/fs/cgroup
tmpfs                  413868       16    413852   1% /run/user/133
tmpfs                  413868       16    413852   1% /run/user/0
/dev/sr0              2839680  2839680         0 100% /media/cdrom0
/dev/mapper/loop0p1    130798     5724    125074   5% /media/root/9981-3E05
```

steev, 2016-02-11:
that shouldn't affect the image, but it does suck that it's sticking around.

Try adding a

sync
sleep 5

before the umount commands at the bottom of the script to see if that helps.  I've never run the scripts inside VMWare, but a sync should tell it to finish writing anything to disk, and then sleep just a bit to make sure.

AaronNGray, 2016-02-15:
Still getting
```
Checking connectivity... done.
del devmap : loop1p2
device-mapper: remove ioctl on loop1p1 failed: Device or resource busy
Removing temporary build files
Generating sha1sum for kali-1.0-utilite.img
```
with
```
# Unmount partitions

sync
sleep 5

umount $bootp
umount $rootp
kpartx -dv $loopdevice
losetup -d $loopdevice
```

steev, 2016-02-15:
Is it possible that the fat partition is being mounted after it's created by... some sort of auto mounter?  Open in Nautilus or some such?  I will see about downloading vmware and grabbing the kali vmware image and replicating, but I can't seem to reproduce it here on non-virtualized.

AaronNGray, 2016-02-15:
I am using normal Kali image not VMWare/Virtual one. Could this be the
issue ?

steev, 2016-02-15:
The reason it isn't unmounting is because some process has that device open still - a sync should flush any buffers that haven't been written, and the sleep should wait 5 seconds before the next step - which is unmounting that partition.  So the key is to figure out what process is keeping it open.  And that tends to be some sort of auto mounter.

AaronNGray, 2016-02-16:
okay

steev, 2018-10-27:
I'm going to close this issue as I haven't heard back and I can't reproduce the issue here.  Please reopen the issue if it is still happening with the latest script!

closed 2018-10-27 by steev