offensive-security/kali-arm-build-scripts #57: apt deb has't been unpacked

crazypenguin, 2016-02-16:
run rpi2.sh
after:
debootstrap --foreign --arch $architecture kali-rolling kali-$architecture http://$mirror/kali
there have apt deb package in var subdirectory,but not been unpacked
so,when chroot in and run "apt-get update" during the rpi2.sh,it will output apt-get not found
please help me.

steev, 2016-02-16:
Can you give the full output of the script when you run it please?

You shouldn't have to chroot in, so I'm not sure what you mean here.

crazypenguin, 2016-02-16:
➜  kali-arm-build-scripts git:(master) ./rpi2.sh 2.0
I: Retrieving Release
I: Retrieving Release.gpg
I: Checking Release signature
I: Valid Release signature (key id 44C6513A8E4FB3D30875F758ED444FF07D8D0BF6)
I: Retrieving Packages
I: Validating Packages
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional required dependencies: adduser dmsetup insserv libapparmor1 libaudit-common libaudit1 libbz2-1.0 libcap2 libcap2-bin libcryptsetup4 libdb5.3 libdebconfclient0 libdevmapper1.02.1 libgcrypt20 libgpg-error0 libkmod2 libncursesw5 libseccomp2 libsemanage-common libsemanage1 libsystemd0 libudev1 libustr-1.0-1 systemd systemd-sysv
I: Found additional base dependencies: libdns-export100 libffi6 libgmp10 libgnutls-openssl27 libgnutls30 libhogweed4 libicu55 libidn11 libisc-export95 libjson-c2 liblz4-1 libmnl0 libnetfilter-acct1 libnettle6 libnfnetlink0 libp11-kit0 libpsl0 libtasn1-6
I: Checking component main on http://http.kali.org/kali...
I: Retrieving adduser 3.113+nmu3
I: Validating adduser 3.113+nmu3
I: Retrieving apt 1.2.3
I: Validating apt 1.2.3
I: Retrieving apt-utils 1.2.3
I: Validating apt-utils 1.2.3
I: Retrieving base-files 1:2016.1.0
I: Validating base-files 1:2016.1.0
I: Retrieving base-passwd 3.5.39
I: Validating base-passwd 3.5.39
I: Retrieving bash 4.3-14+b1
I: Validating bash 4.3-14+b1
I: Retrieving bsdmainutils 9.0.6+b1
I: Validating bsdmainutils 9.0.6+b1
I: Retrieving bsdutils 1:2.27.1-3
I: Validating bsdutils 1:2.27.1-3
I: Retrieving coreutils 8.24-1
I: Validating coreutils 8.24-1
I: Retrieving cpio 2.11+dfsg-4.1
I: Validating cpio 2.11+dfsg-4.1
I: Retrieving cron 3.0pl1-128
I: Validating cron 3.0pl1-128
I: Retrieving dash 0.5.8-2.1
I: Validating dash 0.5.8-2.1
I: Retrieving debconf 1.5.58
I: Validating debconf 1.5.58
I: Retrieving debconf-i18n 1.5.58
I: Validating debconf-i18n 1.5.58
I: Retrieving debian-archive-keyring 2014.3
I: Validating debian-archive-keyring 2014.3
I: Retrieving debianutils 4.7
I: Validating debianutils 4.7
I: Retrieving diffutils 1:3.3-3
I: Validating diffutils 1:3.3-3
I: Retrieving dmidecode 3.0-2
I: Validating dmidecode 3.0-2
I: Retrieving dmsetup 2:1.02.115-2
I: Validating dmsetup 2:1.02.115-2
I: Retrieving dpkg 1.18.4
I: Validating dpkg 1.18.4
I: Retrieving e2fslibs 1.42.13-1
I: Validating e2fslibs 1.42.13-1
I: Retrieving e2fsprogs 1.42.13-1
I: Validating e2fsprogs 1.42.13-1
I: Retrieving emdebian-archive-keyring 2.0.5
I: Validating emdebian-archive-keyring 2.0.5
I: Retrieving findutils 4.6.0+git+20160126-2
I: Validating findutils 4.6.0+git+20160126-2
I: Retrieving gcc-4.8-base 4.8.5-4
I: Validating gcc-4.8-base 4.8.5-4
I: Retrieving gcc-4.9-base 4.9.3-10
I: Validating gcc-4.9-base 4.9.3-10
I: Retrieving gcc-5-base 5.3.1-8
I: Validating gcc-5-base 5.3.1-8
I: Retrieving gnupg 1.4.20-1
I: Validating gnupg 1.4.20-1
I: Retrieving gpgv 1.4.20-1
I: Validating gpgv 1.4.20-1
[... 348 lines cut ...]
I: Extracting ncurses-bin...
I: Extracting passwd...
I: Extracting perl-base...
I: Extracting sed...
I: Extracting sensible-utils...
I: Extracting startpar...
I: Extracting systemd...
I: Extracting systemd-sysv...
I: Extracting sysv-rc...
I: Extracting sysvinit-utils...
I: Extracting tar...
I: Extracting tzdata...
I: Extracting util-linux...
I: Extracting zlib1g...
I: Keyring file not available at /usr/share/keyrings/kali-archive-keyring.gpg; switching to https mirror https://mirrors.kernel.org/debian
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing description
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing architecture
Adding 'local diversion of /usr/sbin/invoke-rc.d to /usr/sbin/invoke-rc.d.chroot'
/third-stage: line 7: apt-get: command not found
/third-stage: line 8: apt-get: command not found
/third-stage: line 12: apt-get: command not found
/third-stage: line 13: apt-get: command not found
/third-stage: line 14: apt-get: command not found

steev, 2016-02-16:
And what is your host machine?  Did you make some modifications to the script at all?

crazypenguin, 2016-02-16:
i am using kali-rolling updated
don't change script at all
and i surely can chroot in,but don't find apt* in any */bin or */sbin directory

steev, 2016-02-16:
I mean what is your host machine - is it a VMWare image, is it an amd64 machine, i386?

crazypenguin, 2016-02-16:
amd64 machine

crazypenguin, 2016-02-16:
it has downloaded apt package:
I: Retrieving apt 1.2.3
I: Validating apt 1.2.3

but don't extract it:
I: Chosen extractor for .deb packages: dpkg-deb
I: Extracting adduser...
I: Extracting base-files...
I: Extracting base-passwd...
I: Extracting bash...

and "debootstrap --foreign --arch $architecture kali-rolling kali-$architecture http://$mirror/kali" control all of them,so i can't really know how it works

steev, 2016-02-16:
it seems as though it's skipping stage2 on your system for some reason (line 49 in the script) - the stage2 is what extracts them all.  Did you run the build-deps.sh file to make sure you have everything installed?

crazypenguin, 2016-02-16:
oh,get it
when run line 49,run "LANG=C chroot kali-$architecture /debootstrap/debootstrap --second-stage" output:
I: Keyring file not available at /usr/share/keyrings/kali-archive-keyring.gpg; switching to https mirror https://mirrors.kernel.org/debian

steev, 2016-02-16:
I haven't seen that here, so I'm not entirely sure what's going on with it.  If you're not making any changes to the script, is there a reason not to use the prebuilt 2.1 image?

crazypenguin, 2016-02-16:
the output is above:
"I: Extracting tzdata...
I: Extracting util-linux...
I: Extracting zlib1g...
I: Keyring file not available at /usr/share/keyrings/kali-archive-keyring.gpg; switching to https mirror https://mirrors.kernel.org/debian
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
missing description
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
missing architecture
Adding 'local diversion of /usr/sbin/invoke-rc.d to /usr/sbin/invoke-rc.d.chroot'
/third-stage: line 7: apt-get: command not found
/third-stage: line 8: apt-get: command not found
/third-stage: line 12: apt-get: command not found
/third-stage: line 13: apt-get: command not found
/third-stage: line 14: apt-get: command not found"

crazypenguin, 2016-02-16:
the prebuilt 2.1 is running on my rpi2
i just want to make a chroot env in my amd64 laptop within kali-rolling
it need some changes to the script?

steev, 2016-02-16:
The script doesn't do that.

What would be fastest would be to grab the sdcard from the rpi2, pop it into your amd64 laptop, and rsync the contents of the second partition (the root partition) to a folder - do it as root so that permissions stay correct. 

Then you'd copy in the qemu-static binary like what is done in the script, and then you'd be able to chroot into it.

crazypenguin, 2016-02-16:
oh no~
but how to use this script?it cann't make chroot env and img on x86 arch?

steev, 2016-02-16:
No, it doesn't.  It wasn't designed to.  It was designed to create an image file that you write to an sdcard and boot the raspberry pi.  

That being said, creating a script of
https://github.com/offensive-security/kali-arm-build-scripts/blob/master/rpi2.sh#L1-L149 
*should* get you a chroot but, you're on your own with that.  It isn't what the scripts were designed to do, and you're better off posting in the forums for support rather than here since this is not an issue with the script.

steev, 2016-02-16:
Upon further research, it would appear that debootstrap is currently broken.  You can try to work around it by following what they say in this bug https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=813232 and I've opened a bug about it in the Kali bug tracker as well https://bugs.kali.org/view.php?id=3074

Another possible workaround is to use qemu-debootstrap in place of debootstrap but I haven't tested that to know what the syntax change would be for the debootstrap line(s).

mchineboy, 2016-02-16:
This affects my builds currently as well. Interestingly enough it worked < 24 hours ago. Something transient in the package updates?

closed 2016-02-17 by crazypenguin