offensive-security/kali-arm-build-scripts #58: i think you hasn't understood mine,please see below

crazypenguin, 2016-02-16:
➜  kali-arm-build-scripts git:(master) ./build-deps.sh
正在读取软件包列表... 完成
正在分析软件包的依赖关系树
正在读取状态信息... 完成
autoconf 已经是最新版 (2.69-9)。
automake 已经是最新版 (1:1.15-3)。
automake 已设置为手动安装。
bc 已经是最新版 (1.06.95-9+b1)。
build-essential 已经是最新版 (11.7)。
build-essential 已设置为手动安装。
cgpt 已经是最新版 (0~R48-7647.B-0kali3)。
curl 已经是最新版 (7.47.0-1)。
curl 已设置为手动安装。
debootstrap 已经是最新版 (1.0.78+kali1)。
device-tree-compiler 已经是最新版 (1.4.0+dfsg-2)。
dosfstools 已经是最新版 (3.0.28-2)。
dosfstools 已设置为手动安装。
e2fsprogs 已经是最新版 (1.42.13-1)。
flex 已经是最新版 (2.5.39-8+b1)。
flex 已设置为手动安装。
git 已经是最新版 (1:2.7.0-1)。
gnupg 已经是最新版 (1.4.20-1)。
libssl-dev 已经是最新版 (1.0.2f-2)。
libssl-dev 已设置为手动安装。
lzma 已经是最新版 (9.22-2)。
lzop 已经是最新版 (1.03-3)。
m4 已经是最新版 (1.4.17-5)。
parted 已经是最新版 (3.2-14)。
pixz 已经是最新版 (1.0.6-2)。
qemu-user-static 已经是最新版 (1:2.5+dfsg-5)。
rsync 已经是最新版 (3.1.1-3)。
rsync 已设置为手动安装。
u-boot-tools 已经是最新版 (2016.01+dfsg1-2)。
vboot-kernel-utils 已经是最新版 (0~R48-7647.B-0kali3)。
vboot-utils 已经是最新版 (0~R48-7647.B-0kali3)。
zip 已经是最新版 (3.0-11)。
zip 已设置为手动安装。
zlib1g-dev 已经是最新版 (1:1.2.8.dfsg-2+b1)。
zlib1g-dev 已设置为手动安装。
将会同时安装下列软件：
  esound-common g++-5-multilib gcc-5-multilib lib32asan2 lib32atomic1
  lib32cilkrts5 lib32gcc-5-dev lib32gcc1 lib32gomp1 lib32itm1 lib32mpx0
  lib32quadmath0 lib32stdc++-5-dev lib32stdc++6 lib32ubsan0 libaudiofile-dev
  libaudiofile1 libbison-dev libc6-dev-i386 libc6-dev-x32 libc6-i386 libc6-x32
  libesd0 libx32asan2 libx32atomic1 libx32cilkrts5 libx32gcc-5-dev libx32gcc1
  libx32gomp1 libx32itm1 libx32quadmath0 libx32stdc++-5-dev libx32stdc++6
  libx32ubsan0
建议安装：
  bison-doc lib32stdc++6-5-dbg libx32stdc++6-5-dbg pulseaudio-esound-compat
  ncurses-doc
下列【新】软件包将被安装：
  abootimg bison btrfs-tools esound-common g++-5-multilib g++-multilib
  gcc-5-multilib gcc-multilib git-core gperf kpartx lib32asan2 lib32atomic1
  lib32cilkrts5 lib32gcc-5-dev lib32gcc1 lib32gomp1 lib32itm1 lib32mpx0
  lib32quadmath0 lib32stdc++-5-dev lib32stdc++6 lib32ubsan0 libaudiofile-dev
  libaudiofile1 libbison-dev libc6-dev-i386 libc6-dev-x32 libc6-i386 libc6-x32
  libesd0 libesd0-dev libncurses5-dev libx32asan2 libx32atomic1 libx32cilkrts5
  libx32gcc-5-dev libx32gcc1 libx32gomp1 libx32itm1 libx32quadmath0
  libx32stdc++-5-dev libx32stdc++6 libx32ubsan0 schedtool
升级了 0 个软件包，新安装了 45 个软件包，要卸载 0 个软件包，有 0 个软件包未被升级。
需要下载 17.8 MB 的归档。
解压缩后会消耗 84.3 MB 的额外空间。
获取:1 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 libaudiofile1 amd64 0.3.6-2+b1 [114 kB]
获取:2 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 esound-common all 0.2.41-11 [45.4 kB]
获取:3 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 libesd0 amd64 0.2.41-11 [17.0 kB]
获取:4 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 libbison-dev amd64 2:3.0.4.dfsg-1 [432 kB]
获取:5 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 bison amd64 2:3.0.4.dfsg-1 [780 kB]
获取:6 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 btrfs-tools amd64 4.4-1 [519 kB]
获取:7 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 libc6-i386 amd64 2.21-7 [2,451 kB]
获取:8 http://mirror.hust.edu.cn/kali kali-rolling/main amd64 libc6-dev-i386 amd64 2.21-7 [1,374 kB]
[... 683 lines cut ...]
Adding 'local diversion of /usr/sbin/invoke-rc.d to /usr/sbin/invoke-rc.d.chroot'
/third-stage: line 7: apt-get: command not found
/third-stage: line 8: apt-get: command not found
/third-stage: line 12: apt-get: command not found
/third-stage: line 13: apt-get: command not found
/third-stage: line 14: apt-get: command not found
chpasswd: (user root) pam_start failure 26
chpasswd: (line 1, user root) password not changed
/bin/sed: can't read /lib/udev/rules.d/75-persistent-net-generator.rules: No such file or directory
/third-stage: line 19: apt-get: command not found

/third-stage: line 20: apt-get: command not found
/third-stage: line 21: apt-get: command not found
Making the image insecure

/bin/sed: can't read /etc/ssh/sshd_config: No such file or directory
update-rc.d: error: cannot find a LSB script for ssh
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing description
dpkg-divert: warning: parsing file '/var/lib/dpkg/status' near line 5 package 'dpkg':
 missing architecture
Removing 'local diversion of /usr/sbin/invoke-rc.d to /usr/sbin/invoke-rc.d.chroot'
/cleanup: line 3: apt-get: command not found
/cleanup: line 4: apt-get: command not found
umount: kali-armhf/proc/sys/fs/binfmt_misc: not mounted

crazypenguin, 2016-02-16:
and i do it follow your readme the same as follow:

mkdir ~/arm-stuff
cd ~/arm-stuff
git clone https://github.com/offensive-security/gcc-arm-linux-gnueabihf-4.7
export PATH=${PATH}:/root/arm-stuff/gcc-arm-linux-gnueabihf-4.7/bin
git clone https://github.com/offensive-security/kali-arm-build-scripts
cd ~/arm-stuff/kali-arm-build-scripts
./build-deps.sh
./rpi2.sh test

steev, 2016-02-16:
Is it possible there is a firewall blocking the https access to mirrors.kernel.org ?  And again, you're better off asking for support in the forums.  Please don't open a third issue about this.

closed 2016-02-16 by steev