offensive-security/kali-arm-build-scripts #59: Beagle Bone Black - Wireless Interface Misnamed because /lib/udev libraries missing

elationfoundation, 2016-02-16:
When I upgraded to the latest rolling release of Kali for beagle bone black the wireless interface started being named `wlx[MAC ADDRESS]`. For instance on my device the wireless interface was named `wlx14cc20187c46`.  Upon finding a [possible solution](https://forums.kali.org/showthread.php?29710-Wireless-interface-name-problem) that stated that there were libraries missing from `/lib/udev`.   When I copied the libraries from the earlier release that were missing in the newer release into the `/lib/udev` folder and re-plugged in the wireless card the interface name correctly set to wlan0.

### Comparison of latest vs. older release files & dirs in /lib/udev

|latest release|Older Release|
|---|---|
||accelerometer|
|ata_id | ata_id |
|cdrom_id|cdrom_id|
|collect|collect|
||dsl-modem.agent|
|hotplug.functions|hotplug.functions|
|hwclock-set|hwclock-set|
|hwdb.bin|hwdb.bin|
|hwdb.d|hwdb.d|
||logger.agent|
|ifupdown-hotplug||
|mtd_probe|mtd_probe|
||net.agent|
||rule_generator.functions|
|rules.d|rules.d|
|scsi_id|scsi_id|
||udev-finish|
|usb_modeswitch|usb_modeswitch|
|v4l_id|v4l_id|
||write_net_rules|

steev, 2016-02-16:
They aren't missing, it's a new version of udev and that's their new naming scheme - if you want to go back to the old way of doing so, either somehow pass net.ifnames=0 to the kernel command line, or you can create an empty file /etc/systemd/network/99-default.link (or symlink it to /dev/null)

See http://www.freedesktop.org/wiki/Software/systemd/PredictableNetworkInterfaceNames

closed 2016-02-16 by steev

elationfoundation, 2016-02-17:
Well so much for all the scripts I have written over the years to find the correct interface number. :sweat_smile: Thanks for this!

strasharo, 2016-03-02:
I proceeded to manually create /etc/udev/rules.d/75-persistent-net-generator.rules with static names for my wireless adapters as a workaround to the new naming scheme. :)