offensive-security/kali-arm-build-scripts #61: bbb, SSH -  Cant login with correct password

0xicl33n, 2016-02-18:
```login as: root
root@10.81.4.30's password: toor
Access denied
root@10.81.4.30's password: toor
Access denied
root@10.81.4.30's password: toor
Access denied
root@10.81.4.30's password: toor
Access denied
root@10.81.4.30's password: toor
Access denied
root@10.81.4.30's password: toor```

```debug2: we did not send a packet, disable method
debug3: authmethod_lookup password
debug3: remaining preferred: ,password
debug3: authmethod_is_enabled password
debug1: Next authentication method: password
root@10.81.4.30's password: toor
debug2: we sent a password packet, wait for reply
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
root@10.81.4.30's password: toor
debug2: we sent a password packet, wait for reply
debug1: Authentications that can continue: publickey,password
Permission denied, please try again.
root@10.81.4.30's password: toor
debug2: we sent a password packet, wait for reply
debug1: Authentications that can continue: publickey,password
debug2: we did not send a packet, disable method
debug1: No more authentication methods to try.
Permission denied (publickey,password).```

I do not understand this anymore lol

steev, 2016-02-18:
Are you sure that 10.81.4.30 is your bbb?

0xicl33n, 2016-02-18:
My router says the hostname of 10.81.4.30 is kali

So yes

steev, 2016-02-18:
I'm not sure - but this is something for either the kali bug system or for the forums.  This isn't an issue with the build scripts.

closed 2016-02-18 by steev

strasharo, 2016-03-02:
I guess this is actually https://github.com/offensive-security/kali-arm-build-scripts/issues/65