offensive-security/kali-arm-build-scripts #62: chromebook-arm-veyron: suitable for Asus C201 ?

duchengyao, 2016-02-23:
`Chromebook C201 Veyron Speedy` is using the same CPU as `Chromebook Flip Veyron Minnie`

I've downloaded `kali-2.1-veyron.img.xz`, and used the dd utility to image this file to my USB key. I can't boot from the USB drive. At the firmware boot screen, after pressing `Control-U',just a beep sound from speaker .

My chromebook is already in developer mode, and dev_boot_usb=1.

steev, 2016-02-23:
How did you use dd on the file to the usb key?

The device tree bindings for both versions of speedy are in the kernel partition, so it should work but I don't have a speedy, so I have no way of testing myself.  Usually a beep on the speaker when hitting ctrl+u means that it doesn't recognize the USB as having a bootable device.

duchengyao, 2016-02-24:
`dd if=kali-2.1-veyron.img of=/dev/XXX bs=1M`

 And I'm trying to follow [Archlinux linstallation](http://archlinuxarm.org/platforms/armv7/rockchip/asus-chromebook-flip-c100p#qt-platform_tabs-ui-tabs2), and just format my SD card follow this command:

         umount /dev/mmcblk1*
         fdisk /dev/mmcblk1
         cgpt create /dev/mmcblk1
         cgpt add -i 1 -t kernel -b 8192 -s 32768 -l Kernel -S 1 -T 5 -P 10 /dev/mmcblk1

After this 4 line command above, I try to Restart and hit ctrl+u. With no beep,some dmsg output, and it seems recognized my SD card as bootable device.

So, I think something wrong with your image format.

Best regards

steev, 2016-02-24:
Hm, if that were the case, then it wouldn't boot on my minnie at all.  It seems like something else is happening.  I'll see about getting a C201 so I can reproduce locally.

duchengyao, 2016-02-25:
After making SD card from kali.img, I repaired the cgpt by following [在三星chromebook安装kali](http://cn.docs.kali.org/armel-armhf-cn/%E5%9C%A8%E4%B8%89%E6%98%9Fchromebook%E5%AE%89%E8%A3%85kali)

         cgpt repair /dev/sdb
         cgpt add -i 1 -t kernel -l Kernel -S 1 -T 5 -P 10 /dev/sdb

Chromebook recognized my SD card, and kali linux has been started.

steev, 2016-02-25:
So after running dd, you run cgpt repair and then it works fine?

duchengyao, 2016-02-25:
yes

duchengyao, 2016-03-09:
duchengyao@kali:~/Downloads$ sudo cgpt repair /dev/sdc
        Secondary Entries is updated.
        Secondary Header is updated.

steev, 2016-03-09:
Okay, I'll try generating a new image and as a last step running cgpt repair on it - someone else asked about the image for the Hisense chromebook and they had to do the same thing.  I don't have to do it with the flip, so it's not easy to track down and that seems like the most logical step.

steev, 2016-03-25:
Can you try the 2.1.2 image?  I run cgpt repair as a last step, see https://github.com/offensive-security/kali-arm-build-scripts/blob/master/chromebook-arm-veyron.sh#L2314

Please let me know if this worked for you or if you have to run it yourself still.

duchengyao, 2016-04-03:
2.1.2 image works well, thanks!

closed 2016-04-03 by steev