offensive-security/kali-arm-build-scripts #65: 2.1 : SSH password login disabled by default on BBB

strasharo, 2016-03-01:
Currently the default sshd setup on BBB contains the following line, which makes remote login impossible:

PermitRootLogin prohibit-password

I checked the build script and it has the following line:
sed -i -e 's/PermitRootLogin without-password/PermitRootLogin yes/' /etc/ssh/sshd_config

I guess it should be updated to:
sed -i -e 's/PermitRootLogin prohibit-password/PermitRootLogin yes/' /etc/ssh/sshd_config

Thanks.

steev, 2016-03-01:
Yep, I noticed that as well - it's a recent update to openssh that changed the default (and option) - will update the scripts.

steev, 2016-03-25:
2.1.2 fixes this issue, thanks for reporting!

closed 2016-03-25 by steev

strasharo, 2016-03-26:
I confirm that it's resolved in 2.1.2, thanks for the help. :)

dimonik, 2017-07-20:
I wrote kali-2017.01-bbb.img to microSD card,
inserted it into BeagleBoneBlack.

Pings are successful, but i can't do SSH.
ssh: connect to host 192.168.1.11 port 22: Connection timed out
sshd_config has PermitRootLogin==Yes.

I don't have keyboard, mouse, monitor. How can I enable SSH?