offensive-security/kali-arm-build-scripts #66: usb0 is not automatically brought up on BBB

strasharo, 2016-03-01:
Interface usb0 is not being automatically brought up on BeagleBone black for me, which makes the initial headless setup a bit complicated. Adding "auto usb0" to /etc/network/interfaces does the trick. I think that will be a good addition to the default setup script.

steev, 2016-03-01:
Good catch.  I've updated the build script and built a new image, as soon as the other images are finished building and tested wrt the ssh issue, they will be posted.

closed 2016-03-01 by steev

strasharo, 2016-03-01:
The version over here still doesn't have the change:
https://github.com/offensive-security/kali-arm-build-scripts/blob/master/bbb.sh#L65

Perhaps it was updated somewhere else?

steev, 2016-03-01:
Correct, I haven't pushed that change yet - once everything is tested locally I push them.

strasharo, 2016-03-01:
Great, thank you. :)