offensive-security/kali-arm-build-scripts #70: net.ifnames=0 causing the device to drop off if networkmanager is used before it's enabled

melikyuksel, 2016-03-15:
Actually, in practice, `ln -s /dev/null /etc/udev/rules.d/80-net-setup-link.rules` caused this, haven't tried it with `net.ifnames=0`. But I imagine the result is the same.

steev sent me.

steev, 2016-03-15:
Not quite; if net.ifnames=0 were used, it wouldn't change until the next reboot. Which is what I intend to do here.

By changing via the symlink, it causes network manager to no longer manage the device, since the device name is no longer en203402x but now eth0, which is listed in /etc/network/interfaces; probably commenting out that eth0 section of the mentioned file would cause network manager to set up the device again.  But it's still easier to just pass the kernel command line option when I build the images.

steev, 2016-03-25:
2.1.2 enables the kernel config option across the boards, and the scripts are updated to show it as well so closing the issue.  Thanks for reporting!

closed 2016-03-25 by steev