offensive-security/kali-arm-build-scripts #78: Weird fatal corruption of OS at random times.

Bry-fi, 2016-05-26:
So I burn the kali ISO (recent version) onto the sdcard. Partition it correctly using gparted. Plug it in to the TV and it boots up fine. Installs work correctly everything is dandy and sunshine. But with no warning, the whole OS fails to boot and just gets stuck at the login screen. The raspberry pi light is red also. I login and nothing pops up so as I did with the old backtrack, I run the command "startx" and it refuses to start up.

steev, 2016-05-26:
Need a bit more information here.

"recent version" isn't a version we've released, please give the actual version you used.

Near the end you've finally mentioned that it's a raspberry pi.  Which version of the raspberry pi?

"i log in and nothing pops up" and "it refuses to start up" aren't very helpful either.

When you say without warning the OS fails to boot; what were you doing before the reboot, how did you shut the machine down.  Did the machine lock up and that's why you restarted it?  Can you provide actual output like the Xorg log, dmesg output, and the output of the "startx" command when you attempt to run it?

When you say it gets "stuck" at the login screen, which?  Console? LightDM?  Please provide a lot more information to debug this (keep in mind that just because it's an ARM device, it doesn't change the troubleshooting steps much at all.

steev, 2017-03-23:
Due to a lack of clarification, I'm going to close the issue.  Please open a new one, or re-open this if you still have issues.

closed 2017-03-23 by steev