offensive-security/kali-arm-build-scripts #80: Odroid-C2 Installation

vladcalina95, 2016-08-23:
Can someone please take me through all the steps of installing kali linux on and odroid-c2 (32GB emmc) once i flash the image to the emmc, i boot once and if i sudo reboot it wont work anymore. I think it has something to do with the volume size and doing the right things with the dd and fdisk tools. Can anyone please be so kind as to take me through all the exact steps of how i should prepare the emmc and how to fully install kali linux. Thank you!

steev, 2016-08-23:
This is a duplicate of issue #76 Please see the instructions in there.  There needs to be space available at the end of the eMMC otherwise it doesn't boot properly.

closed 2016-08-23 by steev

vladcalina95, 2016-08-23:
yes, but i dont understand how to solve the issue from #76 thats why i was asking for help with exact instructions. Thank You!