offensive-security/kali-arm-build-scripts #81: Odroid-C2 Installation

vladcalina95, 2016-08-23:
Can someone please take me through all the steps of installing kali linux on and odroid-c2 (32GB emmc) once i flash the image to the emmc, i boot once and if i sudo reboot it wont work anymore. I think it has something to do with the volume size and doing the right things with the dd and fdisk tools. Can anyone please be so kind as to take me through all the exact steps of how i should prepare the emmc and how to fully install kali linux. Thank you!

vladcalina95, 2016-08-23:
not sure how to use fdsik to allocate everything kali needs and how to allow enough space at the end or something along those lines for it to work (or so i was told)

steev, 2016-08-23:
This issues is for issues with the build scripts.  Please use the kali forums for general linux assistance.

closed 2016-08-23 by steev

vladcalina95, 2016-08-23:
Where is that at? Aslo cant you please help me with the steps of installing it correctly? Also, is microSD easier to install on than emmc?
 
Thank you in advance!

steev, 2016-08-23:
microSD and eMMC are identical.  Please re-read through the thread in issue #76.  I will not reiterate it just because you can't be bothered to read.  If you don't know where the kali linux forums are, I'm sure a search engine could help.  Please stop commenting on closed bugs.

vladcalina95, 2016-08-24:
still cant get kali running after reboot, please help!