offensive-security/kali-arm-build-scripts #82: USB Armory image no longer boots after apt-get upgrade

Glennvd, 2016-09-30:
Trying to install the USB armory image, works great. Until we run an apt-get upgrade and reboot. Armory seems to boot (led goes from dimmed to bright) but the kernel heartbeat is missing. Not really sure how to debug this particular issue. Reimaged twice now in order to confirm.

Anybody have any ideas?

HaydenJayAylor, 2016-09-30:
Did you check the assembly scripts for errors?  Could there be some sort of transmission error when you run apt-get upgrade?  

-Aylor

courtneygibson, 2016-10-13:
I had the same issue.  It turns out the most recent set up updates leaves an empty "modules" file, which prevents loading a number of key modules on boot.  Adding the following lines to the bottom of **/etc/modules** seems to fix the problem:

```
ledtrig_heartbeat
ci_hdrc_imx
#g_mass_storage
g_ether
```

closed 2017-10-02 by steev