offensive-security/kali-arm-build-scripts #84: RPI2/3 add a patch to enable monitor mode for RTL8188CUS via USB (edimax small dongle)

gilsport, 2016-10-29:
hi,
there are plenty of guides how to do it manually.
and since everyone with a rpi and an Edimax dongle probably would like to have monitor mode,
can you please add it as a patch to the rpi2.sh script ? just like the injection script.
a guide can be found here

http://raspberrypi.stackexchange.com/questions/36747/enable-monitoring-mode-for-rtl8188cus-via-usb-on-raspbian

thanks,
appreciate all your work!

steev, 2016-12-21:
To be frank, that dongle is terrible, and the rtlwifi driver isn't as stable for it; so the question becomes one of, would you prefer a more stable driver, or working monitor mode?  Why not both and use a better dongle?

I'll look into this but I'm not guaranteeing it will change.  It will have to depend on whether or not the 4.4 driver is better.

steev, 2017-10-02:
Can you test with the 2017.2 image and tell me if it works?

closed 2017-10-02 by steev