offensive-security/kali-arm-build-scripts #87: orange pi zero

vay3t, 2016-12-26:
Even though it is not very well known in the market I think that having kali in "orange pi zero" would be very useful, "orange pi zero" is a replacement for "raspberry pi zero" (exhausted and hard to get). "Orange pi zero" is more accessible to the general public from "aliexpress"

Ty6836, 2017-02-02:
Please create Kali for Orange Pi Zero

sadk, 2017-02-28:
Please create Kali for Orange Pi Zero / nanopi neo aire  + 1

DavidSin02, 2019-03-25:
Please support this device!

DavidSin02, 2019-03-25:
**Orange Pi Zero Hardware Details**

- http://linux-sunxi.org/Xunlong_Orange_Pi_Zero

- http://www.orangepi.org/orangepizero/

**Orange Pi Zero Kernel Source Code and Build**

- https://github.com/orangepi-xunlong/OrangePi_Build

- https://mega.nz/#F!MKQSUIbS!reCl8EK0QqjnOoC-e2ZwBg

DavidSin02, 2019-03-25:
I hope it helps