offensive-security/kali-arm-build-scripts #95: usbarmory wrong dts files

scaery, 2017-03-13:
PLS UP!

while building u will c what i am referring... have a look here!

`https://github.com/inversepath/usbarmory/tree/master/software/kernel_conf`

the old files remain in dts-4.8

example:
`wget https://raw.githubusercontent.com/inversepath/usbarmory/master/software/kernel_conf/dts-4.8/imx53-usbarmory-common.dtsi -O arch/arm/boot/dts/imx53-usbarmory-common.dtsi`

btw, pl0x up the u-boot  too ;)

wget ftp://ftp.denx.de/pub/u-boot/u-boot-2017.03.tar.bz2

steev, 2017-03-22:
Yep, will be pushing the latest soon (tm), moving to 4.10 kernel, and 2017.01 (not 03 yet)

scaery, 2017-08-20:
This is my last working patch. 

```
--- usbarmory.sh	2017-06-24 23:47:47.128975637 +0200
+++ usbarmory-patch.sh	2017-08-21 00:39:03.931912279 +0200
@@ -162,9 +162,9 @@
 
 # Create the disk and partition it
 echo "Creating image file for USB Armory"
-dd if=/dev/zero of=${basedir}/kali-$1-usbarmory.img bs=1M count=7000
+fallocate -l 3500MiB ${basedir}/kali-$1-usbarmory.img
 parted kali-$1-usbarmory.img --script -- mklabel msdos
-parted kali-$1-usbarmory.img --script -- mkpart primary ext2 5M 100%
+parted kali-$1-usbarmory.img --script -- mkpart primary ext4 5M 100%
 
 # Set the partition variables
 loopdevice=`losetup -f --show ${basedir}/kali-$1-usbarmory.img`
@@ -174,7 +174,7 @@
 rootp=${device}p1
 
 # Create file systems
-mkfs.ext2 $rootp
+mkfs.ext4 -F $rootp
 
 # Create the dirs for the partitions and mount them
 mkdir -p ${basedir}/root
@@ -212,9 +212,9 @@
 
 allow-hotplug usb0
 iface usb0 inet static
-address 10.0.0.1
+address 10.10.10.1
 netmask 255.255.255.0
-gateway 10.0.0.2
+gateway 10.10.10.2
 EOF
 
 cat << EOF > ${basedir}/root/etc/apt/sources.list
@@ -251,8 +251,8 @@
 log-facility local7;
 
 # A slightly different configuration for an internal subnet.
-subnet 10.0.0.0 netmask 255.255.255.0 {
-  range 10.0.0.2 10.0.0.2;
+subnet 10.10.10.0 netmask 255.255.255.0 {
+  range 10.10.10.2 10.10.10.2;
   default-lease-time 600;
   max-lease-time 7200;
 }
@@ -349,42 +349,37 @@
 
 # Kernel section. If you want to use a custom kernel, or configuration, replace
 # them in this section.
-git clone -b linux-4.2.y --depth 1 git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git ${basedir}/root/usr/src/kernel
+git clone -b linux-4.10.y --depth 1 git://git.kernel.org/pub/scm/linux/kernel/git/stable/linux-stable.git ${basedir}/root/usr/src/kernel
 cd ${basedir}/root/usr/src/kernel
 git rev-parse HEAD > ../kernel-at-commit
 touch .scmversion
 export ARCH=arm
 export CROSS_COMPILE=arm-linux-gnueabihf-
-patch -p1 --no-backup-if-mismatch < ${basedir}/../patches/mac80211.patch
-wget https://raw.githubusercontent.com/inversepath/usbarmory/master/software/kernel_conf/imx53-usbarmory-common.dtsi -O arch/arm/boot/dts/imx53-usbarmory-common.dtsi
+# fix kali wifi patches for 4.10.y // TODO patch tx.c
+mkdir ${basedir}/patches
+cp ${basedir}/../patches/kali-wifi-injection-4.4.patch ${basedir}/patches/kali-wifi-injection-4.10.patch
+sed -i '13,48d' ${basedir}/patches/kali-wifi-injection-4.10.patch
+sed -i 's/wireless\/zd1211rw/wireless\/zydas\/zd1211rw/g' ${basedir}/patches/kali-wifi-injection-4.10.patch
+patch -p1 --no-backup-if-mismatch < ${basedir}/patches/kali-wifi-injection-4.10.patch
+#rm -rf ${basedir}/patches/kali-wifi-injection-4.10.patch
 wget https://raw.githubusercontent.com/inversepath/usbarmory/master/software/kernel_conf/imx53-usbarmory-host.dts -O arch/arm/boot/dts/imx53-usbarmory-host.dts
[... 166 lines cut ...]
 	flags |= ieee80211_get_tx_rate(dev, info)->hw_value << 24;
+
+	// When this flag is set the firmware waits untill ALL fragments have
+	// reached the USB device. Then it sends the first fragment and waits
+	// for ACKS's. Of course in monitor mode it won't detect these ACK's.
 	if (ieee80211_has_morefrags(tx_hdr->frame_control))
-		flags |= RTL818X_TX_DESC_FLAG_MOREFRAG;
+	{
+		// If info->control.vif is NULL it's most likely in monitor mode
+		if (likely(info->control.vif != NULL && info->control.vif->type != NL80211_IFTYPE_MONITOR)) {
+			flags |= RTL818X_TX_DESC_FLAG_MOREFRAG;
+		}
+	}
 
 	/* HW will perform RTS-CTS when only RTS flags is set.
 	 * HW will perform CTS-to-self when both RTS and CTS flags are set.
```

Finally just...

patch -p1 usbarmory.sh < patch

and build usbarmory.sh as usual.

Please push some love into your repo as this issue idles for so long!

closed 2017-10-02 by scaery