offensive-security/kali-arm-build-scripts #100: chromebook-veyron builds but will not boot "kernel.bin: No such file or directory"

SolidHal, 2017-03-30:
HostOS: Kali Linux latest (ran apt-get update, upgrade, and dist-upgrade)

Possibly related Errors: 
> I: Keyring file not available at /usr/share/keyrings/kali-archive-keyring.gpg; switching to https mirror https://deb.debian.org/debian

> dd: failed to open '/root/arm-stuff/kali-arm-build-scripts/veyron-2.12/kernel.bin': No such file or directory

After writing the image to usb, the chromebook refuses to boot from it. I have booted Archlinux successfully. 

[veyron_log.txt](https://github.com/offensive-security/kali-arm-build-scripts/files/880980/veyron_log.txt)

steev, 2017-03-30:
The image won't fail building because of the kernel failing to build, but as you noticed, the kernel didn't successfully compile (so no, the image won't boot, because there is no kernel to build)

That said, it looks like they've done something with a back port that breaks POSIX_ACL in the kernel and until that's fixed, the image won't boot properly.  I'm currently working on this and hope to have a fix pushed soon.

SolidHal, 2017-03-31:
That makes sense. Would using an older release of kali work as a temporary workaround? 
Also, does it not cause issues with building if it choose to get some of the packages from the debian mirror instead of the kali mirror? I feel like I had read somewhere that it doesn't work properly if the kali mirror isn't used.

SolidHal, 2017-03-31:
Theres also this string of errors from the log that may or may not be useful:
> Setting up fuse (2.9.7-1) ...
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/cpuid/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/fuse/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/nfnetlink_queue/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/nfnetlink_log/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/nfnetlink/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/bluetooth/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/rfkill/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/ppdev/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/joydev/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/intel_powerclamp/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/intel_rapl_perf/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/evdev/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/pcspkr/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/serio_raw/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/sg/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/binfmt_misc/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/snd_intel8x0/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/battery/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/parport_pc/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/parport/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/video/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/ac/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/snd_ac97_codec/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/button/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/ac97_bus/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/snd_pcm/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/snd_timer/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/snd/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/soundcore/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/acpi_cpufreq/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/tpm_tis/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/tpm_tis_core/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/tpm/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/ip_tables/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/x_tables/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/autofs4/holders': No such file or directory
[... cut ...]
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/ata_piix/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/e1000/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/libata/holders': No such file or directory
> libkmod: ERROR ../libkmod/libkmod-module.c:1929 kmod_module_get_holders: could not open '/sys/module/scsi_mod/holders': No such file or directory

steev, 2017-03-31:
it can be ignored.

steev, 2017-03-31:
The kernel build has nothing to do with kali v debian repositories.  Upstream (Chrome/Chromium) kernel devs did something very stupid, and broke POSIX ACL pulling in some things to support Android on ChromeOS. 

The immediate fix is to edit the chromebook-rockchip... file in kernel-configs and set the line that has (roughly) 'CONFIG_EXT4_POSIX_ACL=y' to be '# CONFIG_EXT4_POSIX_ACL is not set'

I've only tested that it compiles, not any Kali interaction just yet, I'm currently building the image to test it.

You can definitely still use an older Kali image, and just update the software via apt update && apt dist-upgrade, it will just take a long time as there are a lot of packages that have been updated/changed since the last release.

tvondra, 2017-04-17:
I've managed to get the veyron images working, after a bit of struggle. I know @steev probably already fixes most of the issues in his repository, so I'm not going to send pull requests - if needed, look at this branch: https://github.com/tvondra/kali-arm-build-scripts/tree/fixes

In general, it works. One thing I ran into is that after restart I have to reinstall qemu-user-static package, otherwise I get "Exec format error" when the shell script runs debootstrap. Apparently the qemu-arm gets empty or something.

I intend to use the image on my C201PA chromebook, and while I haven't done much testing yet the image seems to be running OK, with one exception - logging out from from xfce freezes the screen for a while. I assume it's something about the GPU, perhaps.

One change that is not merely a fix is enabling all available encryption algorithms in kernel. That's pretty much the reason why I started looking into building custom images, because I want/need to use LUKS, and that doesn't work with the default kernel.

@steev, Let me know if you're OK with adding this to the official image, and if you need a pull request.

steev, 2017-04-19:
I've seen the same issue, and it's definitely kernel side.

I looked through some of your changes, and they seem to be okay - I've actually already integrated some of the config changes (although I may not enable everything that you have) - once I finish what I'm working on and push it, please take a look and compare to yours.  Pull requests are always appreciated though!

tvondra, 2017-08-09:
Hi, any progress on this? Is there a way to help moving it forward?

steev, 2017-10-02:
This is working now - the issue was (and is) upstream's kernel support for Android breaks ACLs for ext4.  The only solution is to disable them, which is done in the latest image.

closed 2017-10-02 by steev