offensive-security/kali-arm-build-scripts #105: Serial connection - login & pass incorrect | bbb image 2017.01

dimonik, 2017-07-22:
I wrote 
[kali-2017.01-bbb.img](https://images.offensive-security.com/arm-images/kali-2017.01-bbb.img.xz)
 to microSD card,
inserted it into BeagleBoneBlack.

Using FT232RL USB<->Serial board, i managed to connect to BeagleBoneBlack.

I see output like this:
```
Debian GNU/Linux 8 beagle ttyS0

rcn-ee.net console Debian Image 2016-06-09

Support/FAQ: http://elinux.org/BeagleBoardDebian

default username:password is [debian:temppwd]

The IP Address for usb0 is: 192.168.7.2
beagle login: 
```
but DEFAULT Kali login=**root** and password=**toor** are not working!

What are correct credentials?

dimonik, 2017-07-22:
I found the reason: corrupted micro SD card. So it is hardware error.

dimonik, 2017-08-05:
I continued to dig to the roots of the problem.

It is NOT corruption of SD card.
BBB can't read microSD ONLY after writing [2017.01 image](https://images.offensive-security.com/arm-images/kali-2017.01-bbb.img.xz)

I tried both commands:
sudo dd bs=512k if=kali-2017.01-bbb.img | pv | sudo dd of=/dev/mmcblk0
sudo dd bs=4M if=kali-2017.01-bbb.img | pv | sudo dd of=/dev/mmcblk0

And with both of them the result is:
`U-Boot> mmc info`
`Card did not respond to voltage select!`

While the same card formatted with one fat32 partition respond as:
`U-Boot> mmc info`
`Device: OMAP SD/MMC`
`Manufacturer: ID: 27`
`OEM: 5048`
`etc...`

dimonik, 2017-08-06:
The reason is that image can't be read after written by dd to SD card.

This is my workaround:
- still use dd to write image to SD card;
- after, re-insert this SD into the laptop, usually laptop easily read such cards despite that BBB can't;
- make two directories on your laptop: part1 and part2;
- copy everything from SD card's smaller volume into directory part1
- copy everything from SD card's larger volume into directory part2
- start Gparted utility on your laptop;
- using Gparted delete from SD card 1st and 2nd partition;
- open menu Device, choose Create Partition Table... select ms-dos type and OK
- open menu Partition, choose New, create new partition (128 MiB, fat16, label=p1);
- after, create another partition the same way but with this params (size up to the end, ext4, label=p2);
- copy everything from part1 to the partition labeled as p1;
- copy everything from part2 to the partition labeled as p2;
- unmount p1 and p2 from your OS;
- eject SD card.

Now you can use this card to boot BBB successfully!

steev, 2018-10-27:
Are you still having this issue?  I can't reproduce here at all to have to essentially remake the sdcard, so I'm not entirely sure what the difference is.  What sdcard are you using?