offensive-security/kali-arm-build-scripts #106: Original (Marvell based) CuBox not booting

6arms1leg, 2017-08-02:
My old original (Marvell based) CuBox is not booting the current [Kali Linux image](https://images.offensive-security.com/arm-images/kali-2017.01-cubox.img.xz) from the [download page](https://docs.kali.org/kali-on-arm/install-kali-arm-on-a-cubox).

I'm not an expert, but there is no `uImage` in `/boot/`.
Also, I think there is a typo in the the [build script](kali-arm-build-scripts/cubox.sh) in line [224](https://github.com/offensive-security/kali-arm-build-scripts/blob/master/cubox.sh#L224) and [225](https://github.com/offensive-security/kali-arm-build-scripts/blob/master/cubox.sh#L225): The default load address on the CuBox is `0x02000000`, not `0x00200000` (the `2` should be shifted one to the left). 

Best regards

steev, 2017-09-28:
Does this still occur with 2017.2?

6arms1leg, 2017-10-07:
Yes, it still does not boot with 2017.2. But the `/boot/uImage` file is there now. 👍 
Here is a [log](https://github.com/offensive-security/kali-arm-build-scripts/files/1364659/screenlog.txt) of the failing boot process. I have the correct uBoot version (5.4.4 NQ SR1) installed.
I think line 106 in the log should have succeeded, but for some reason it fails:
```
===> Executing ext4load mmc 0:1 0x02000000 /boot/boot.scr
Loading file "/boot/boot.scr" from mmc device 0:1 mmcda1
ext4fs_devread read outside partition 2466988952
** File not found /boot/boot.scr
```
I also tried manually loading the `uImage` from uBoot, which also fails:
```
CuBox>> ext4load mmc 0:1 0x02000000 /boot/uImage
Loading file "/boot/uImage" from mmc device 0:1 mmcda1
ext4fs_devread read outside partition 2466988952
** File not found /boot/uImage
```
The CuBox itself is OK, as I can boot other GNU/Linux distributions (e. g. Arch Linux ARM).

steev, 2017-10-07:
Okay after digging around a bit, I think we may be running into an issue with the latest e2fsprogs.

https://unix.stackexchange.com/questions/304158/u-boot-2016-07-ext4fs-devread-read-outside-partition

So I'll generate a new image that disables that as well.

steev, 2017-10-09:
There should be a new image on the site soon (2017.2.1) - please give it a try and let me know!

steev, 2017-10-19:
It should be posted, can you give it a whirl and let me know?

6arms1leg, 2017-10-19:
Yes, I already noticed and tested it right away, but didn't have time to reply here yet – sorry!
I can confirm that the latest build works, many thanks!
My PR #107 might be irrelevant then (strange, because I had another GNU/Linux image that would only boot from `0x02000000`). Feel free to close it.

Thanks again for your great work!

closed 2017-10-19 by 6arms1leg