offensive-security/kali-arm-build-scripts #111: build kernel with DEVMEM_STRICT=n?

hanetzer, 2017-10-05:
My use case: I want to try to use dmidecode to grab the spd values for the ram on my chromebook (veyron-speedy) for use in u-boot, but due to the named config value being set to y it won't work.
Would be a simple change, and I think others may have use for this in REing some bits of the chromebook.

steev, 2017-10-07:
We will not do that by default, but you're more than welcome to modify the config yourself and build an image.  That's what the scripts are here for :)

closed 2017-10-07 by steev

hanetzer, 2017-10-08:
@steev would gladly, but apparently even installing debootstrap on my gentoo system is not enough to allow this to work, since they call it with some kali specific parameters which config files I don't have.

steev, 2017-10-08:
The only change you should have to make is to create a symlink in /usr/share/debootstrap/scripts 

Specifically create one called kali-rolling and symlink it to /usr/share/debootstrap/scripts/sid

I do this on my gentoo system all the time (I'm not just a kali dev, I'm also a gentoo dev) - the other change would be to rename the qemu binary to the gentoo binary name - qemu-arm-static instead of qemu-static-arm.

hanetzer, 2017-10-08:
Ah, thanks for that. Yeah, I've noticed that as well, cool I should be fine now. Quick question: the resultant image just requires dev mode turned on and usb boot enabled for the chromebook in question?

steev, 2017-10-08:
Yep, that'll take about 15 minutes or so as switching to dev mode wipes the system of anything you have installed.

hanetzer, 2017-10-08:
Aware, its already in dev mode (kinda; its bricked atm because I'm working on u-boot for it so I can hopefully avoid some of the more arcane chromeos stuffs, but once I reflash the stock coreboot its already in dev mode), thanks for the info.