offensive-security/kali-arm-build-scripts #115: rpi0w Cross Compile: Multiple Failures

strixdio, 2017-12-07:
It's probably something I did wrong, sorry if that's the case. Please let me know if you need any more info.

## Build Log:

```
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx)
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional required dependencies: fdisk libaudit-common libaudit1 libbz2-1.0 libcap-ng0 libdb5.3 libdebconfclient0 libgcrypt20 libgpg-error0 liblz4-1 libncursesw5 libsemanage-common libsemanage1 libsystemd0 libtext-charwidth-perl libudev1 
I: Found additional base dependencies: dirmngr dmsetup gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm libapparmor1 libassuan0 libbsd0 libcap2 libcryptsetup4 libdevmapper1.02.1 libdns-export190 libelf1 libfastjson4 libffi6 libgmp10 libgnutls30 libhogweed4 libidn11 libidn2-0 libip4tc0 libip6tc0 libiptc0 libisc-export189 libksba8 libldap-2.4-2 libldap-common liblocale-gettext-perl liblognorm5 libmnl0 libncurses5 libnetfilter-conntrack3 libnettle6 libnfnetlink0 libnpth0 libp11-kit0 libpsl5 libsasl2-2 libsasl2-modules-db libseccomp2 libsqlite3-0 libtasn1-6 libtext-iconv-perl libunistring2 libxtables12 pinentry-curses xxd 
I: Checking component main on http://http.kali.org/kali...
I: Retrieving adduser 3.116
I: Validating adduser 3.116
I: Retrieving apt 1.6~alpha5
I: Validating apt 1.6~alpha5
I: Retrieving apt-utils 1.6~alpha5
I: Validating apt-utils 1.6~alpha5
I: Retrieving base-files 1:2017.3.0
I: Validating base-files 1:2017.3.0
I: Retrieving base-passwd 3.5.44
I: Validating base-passwd 3.5.44
I: Retrieving bash 4.4-5
I: Validating bash 4.4-5
I: Retrieving bsdmainutils 9.0.14
I: Validating bsdmainutils 9.0.14
I: Retrieving bsdutils 1:2.30.2-0.1
I: Validating bsdutils 1:2.30.2-0.1
I: Retrieving coreutils 8.28-1
I: Validating coreutils 8.28-1
I: Retrieving cpio 2.11+dfsg-6
I: Validating cpio 2.11+dfsg-6
I: Retrieving cron 3.0pl1-128.1
I: Validating cron 3.0pl1-128.1
I: Retrieving dash 0.5.8-2.5
I: Validating dash 0.5.8-2.5
I: Retrieving debconf 1.5.65
I: Validating debconf 1.5.65
I: Retrieving debconf-i18n 1.5.65
I: Validating debconf-i18n 1.5.65
I: Retrieving debian-archive-keyring 2017.6
I: Validating debian-archive-keyring 2017.6
I: Retrieving debianutils 4.8.3
I: Validating debianutils 4.8.3
I: Retrieving diffutils 1:3.6-1
I: Validating diffutils 1:3.6-1
I: Retrieving dirmngr 2.2.2-1
I: Validating dirmngr 2.2.2-1
I: Retrieving dmsetup 2:1.02.145-4.1
I: Validating dmsetup 2:1.02.145-4.1
I: Retrieving dpkg 1.19.0.4kali1
I: Validating dpkg 1.19.0.4kali1
I: Retrieving e2fslibs 1.43.7-1
I: Validating e2fslibs 1.43.7-1
I: Retrieving e2fsprogs 1.43.7-1
I: Validating e2fsprogs 1.43.7-1
I: Retrieving fdisk 2.30.2-0.1
I: Validating fdisk 2.30.2-0.1
I: Retrieving findutils 4.6.0+git+20170828-2
I: Validating findutils 4.6.0+git+20170828-2
I: Retrieving gcc-6-base 6.4.0-10
I: Validating gcc-6-base 6.4.0-10
I: Retrieving gcc-7-base 7.2.0-16
I: Validating gcc-7-base 7.2.0-16
I: Retrieving gnupg 2.2.2-1
I: Validating gnupg 2.2.2-1
I: Retrieving gnupg-l10n 2.2.2-1
[... 1508 lines cut ...]
dpkg: regarding .../dpkg_1.19.0.4kali1_armel.deb containing dpkg, pre-dependency problem:
 dpkg pre-depends on libbz2-1.0
  libbz2-1.0:armel is unpacked, but has never been configured.

dpkg: warning: ignoring pre-dependency problem!
dpkg: regarding .../dpkg_1.19.0.4kali1_armel.deb containing dpkg, pre-dependency problem:
 dpkg pre-depends on liblzma5 (>= 5.2.2)
  liblzma5:armel is unpacked, but has never been configured.

dpkg: warning: ignoring pre-dependency problem!
dpkg: regarding .../dpkg_1.19.0.4kali1_armel.deb containing dpkg, pre-dependency problem:
 dpkg pre-depends on libselinux1 (>= 2.3)
  libselinux1:armel is unpacked, but has never been configured.

dpkg: warning: ignoring pre-dependency problem!
dpkg: regarding .../dpkg_1.19.0.4kali1_armel.deb containing dpkg, pre-dependency problem:
 dpkg pre-depends on zlib1g (>= 1:1.1.4)
  zlib1g:armel is unpacked, but has never been configured.

dpkg: warning: ignoring pre-dependency problem!
Preparing to unpack .../dpkg_1.19.0.4kali1_armel.deb ...
Unpacking dpkg (1.19.0.4kali1) over (1.19.0.4kali1) ...
Preparing to unpack .../e2fslibs_1.43.7-1_armel.deb ...
Unpacking e2fslibs:armel (1.43.7-1) over (1.43.7-1) ...
```

steev, 2017-12-08:
I'm not 100% sure as I've not seen that issue before - did you run the build-deps.sh script before attempting to run the rpi0w script?

I'm technically on vacation at the moment and was unable to ssh into my home network, so I can't currently replicate the issue.

strixdio, 2017-12-08:
Yes, I've run build-deps.sh before. This was on a fresh kali x86_64 install.

strixdio, 2017-12-10:
A friend pointed out that I was missing some packages (I swear I thought I had them installed!), so I tried to install th and many errors went away. There are still some errors that we're not sure what to do with:

```
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id xxxxxxxx)
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Found additional required dependencies: fdisk libaudit-common libaudit1 libbz2-1.0 libcap-ng0 libdb5.3 libdebconfclient0 libgcrypt20 libgpg-error0 liblz4-1 libncursesw5 libsemanage-common libsemanage1 libsystemd0 libtext-charwidth-perl libudev1 
I: Found additional base dependencies: dirmngr dmsetup gnupg-l10n gnupg-utils gpg gpg-agent gpg-wks-client gpg-wks-server gpgconf gpgsm libapparmor1 libassuan0 libbsd0 libcap2 libcryptsetup4 libdevmapper1.02.1 libdns-export190 libelf1 libfastjson4 libffi6 libgmp10 libgnutls30 libhogweed4 libidn11 libidn2-0 libip4tc0 libip6tc0 libiptc0 libisc-export189 libksba8 libldap-2.4-2 libldap-common liblocale-gettext-perl liblognorm5 libmnl0 libncurses5 libnetfilter-conntrack3 libnettle6 libnfnetlink0 libnpth0 libp11-kit0 libpsl5 libsasl2-2 libsasl2-modules-db libseccomp2 libsqlite3-0 libtasn1-6 libtext-iconv-perl libunistring2 libxtables12 pinentry-curses xxd 
I: Checking component main on http://http.kali.org/kali...
I: Validating adduser 3.116
I: Validating apt 1.6~alpha5
I: Validating apt-utils 1.6~alpha5
I: Validating base-files 1:2017.3.0
I: Validating base-passwd 3.5.44
I: Validating bash 4.4-5
I: Validating bsdmainutils 9.0.14
I: Validating bsdutils 1:2.30.2-0.1
I: Validating coreutils 8.28-1
I: Validating cpio 2.12+dfsg-6
I: Validating cron 3.0pl1-128.1
I: Validating dash 0.5.8-2.5
I: Validating debconf 1.5.65
I: Validating debconf-i18n 1.5.65
I: Validating debian-archive-keyring 2017.6
I: Validating debianutils 4.8.3
I: Validating diffutils 1:3.6-1
I: Validating dirmngr 2.2.3-1
I: Validating dmsetup 2:1.02.145-4.1
I: Validating dpkg 1.19.0.4kali1
I: Validating e2fslibs 1.43.7-1
I: Validating e2fsprogs 1.43.7-1
I: Validating fdisk 2.30.2-0.1
I: Validating findutils 4.6.0+git+20170828-2
I: Validating gcc-6-base 6.4.0-10
I: Validating gcc-7-base 7.2.0-16
I: Validating gnupg 2.2.3-1
I: Validating gnupg-l10n 2.2.3-1
I: Validating gnupg-utils 2.2.3-1
I: Validating gpg 2.2.3-1
I: Validating gpg-agent 2.2.3-1
I: Validating gpg-wks-client 2.2.3-1
I: Validating gpg-wks-server 2.2.3-1
I: Validating gpgconf 2.2.3-1
I: Validating gpgsm 2.2.3-1
I: Validating gpgv 2.2.3-1
I: Validating grep 3.1-2
I: Validating gzip 1.6-5+b1
I: Validating hostname 3.18+b1
I: Validating ifupdown 0.8.29
I: Validating init 1.51kali1
I: Validating init-system-helpers 1.51kali1
I: Validating iproute2 4.9.0-2.1
I: Validating iptables 1.6.1-2+b1
I: Validating iputils-ping 3:20161105-1
I: Validating isc-dhcp-client 4.3.5-3+b1
I: Validating isc-dhcp-common 4.3.5-3+b1
I: Validating kali-archive-keyring 2015.2
I: Validating kmod 24-1
I: Validating libacl1 2.2.52-3+b1
I: Validating libapparmor1 2.11.1-3
I: Validating libapt-inst2.0 1.6~alpha5
I: Validating libapt-pkg5.0 1.6~alpha5
I: Validating libassuan0 2.4.5-1
I: Validating libattr1 1:2.4.47-2+b2
I: Validating libaudit-common 1:2.8.1-2
I: Validating libaudit1 1:2.8.1-2
I: Validating libblkid1 2.30.2-0.1
[... 3302 lines cut ...]
  HOSTCC  firmware/ihex2fw
  IHEX2FW firmware/emi26/loader.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi26/loader.fw
  IHEX2FW firmware/emi26/firmware.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi26/firmware.fw
  IHEX2FW firmware/emi26/bitstream.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi26/bitstream.fw
  IHEX2FW firmware/emi62/loader.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi62/loader.fw
  IHEX2FW firmware/emi62/bitstream.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi62/bitstream.fw
  IHEX2FW firmware/emi62/spdif.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi62/spdif.fw
  IHEX2FW firmware/emi62/midi.fw
  INSTALL /git/kali-arm-build-scripts/rpi0w-nexmon-p4wnp1-1.0/root/lib/firmware/emi62/midi.fw
  IHEX    firmware/kaweth/new_code.bin
firmware/Makefile:194: recipe for target 'firmware/kaweth/new_code.bin' failed
Makefile:1122: recipe for target 'firmware_install' failed
del devmap : loop0p2
del devmap : loop0p1
Cleaning up the temporary build files...
Generating sha256sum for kali-1.0-rpi0w-nexmon.img
Compressing kali-1.0-rpi0w-nexmon.img
Generating sha256sum for kali-1.0-rpi0w-nexmon.img.xz
```

steev, 2018-03-07:
How much free space is on the machine?  I've never seen the kernel fail to build like that, aside from possibly running out of space.  Also how much memory is available on the machine?

steev, 2018-10-27:
Closing due to lack of response, I can't reproduce the issue here.

closed 2018-10-27 by steev