offensive-security/kali-arm-build-scripts #117: Can't Update Kernel and Stuck at 3.14.79.

jsl303, 2017-12-30:
I'm having a problem updating the kernel. Running uname -r says 3.14.79 after the update.
I downloaded kali-2017.3-odroidc2.img for Odroid c2 from the web page below.
https://www.offensive-security.com/kali-linux-arm-images/
I'm trying to get ALFA AWUS036ACH with RTL8812AU chip to work.
I tried to run the commands explained in Kali web page.
https://www.kali.org/news/kali-linux-20171-release/
apt install realtek-rtl88xxau-dkms
Building initial module for 3.14.79
Error! Bad return status for module build on kernel: 3.14.79 (aarch64)
/var/lib/dkms/realtek-rtl88xxau/5.1.5~20171206/build/make.log
                                                                               Makefile:590: /usr/src/kernel/arch/aarch64/Makefile: No such file or directory
make[1]: *** No rule to make target '/usr/src/kernel/arch/aarch64/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1857: recipe for target 'modules' failed
make: *** [modules] Error 2

I tried to update by running apt install linux-headers-arm64 linux-image-arm64.
It complained that it can't find some files in /boot, so I mounted the boot volume, and ran
mount /dev/mmcblk0p1 /boot
dpkg --configure --a
I got no error except
ln: failed to create hard link '/boot/initrd.img-4.14.0-kali1-arm64.dpkg-bak' => '/boot/initrd.img-4.14.0-kali1-arm64': Operation not permitted
I googled, and it seems like people are saying you can ignore those.
It seemed all the updated files were in /boot.
Image                                                                           
System.map-4.14.0-kali1-arm64                                                   
boot.ini                                                                        
config-4.14.0-kali1-arm64                                                       
initrd.img-4.14.0-kali1-arm64                                                   
meson64_odroidc2.dtb                                                            
mkuinitrd                                                                       
uInitrd                                                                         
vmlinuz-4.14.0-kali1-arm64                                                      
However when I reboot and run uname-r, it still says 3.14.79.
How can I update the kernel, so I can install realtek-rtl88xxau-dkms?
Thanks!

muddy-13, 2018-02-28:
I was getting the same error and when I tried : 
make mr proper
I got: 
make: *** No rule to make target 'mr'. stop

steev, 2018-03-01:
you need to install "bc" - but also, it should be make ARCH=arm64 not ARCH=aarch64 (if you're not sure, you can look in /usr/src/kernel/arch/ for valid ARCHs.

muddy-13, 2018-03-01:
Thanks Steev - I'll look into this tomorrow

muddy-13, 2018-03-01:
Hi Steve,

 I downloaded the bc module - but I am unsure where to make the edits you recommended in the ARCH=aarch64      to ARCH=arm64 - I can't find the reference in a makefile.

/usr/src/kernel/arch does show a directory arm64

but grep -r ARCH=aarch64        comes up blank

steev, 2018-03-03:
I'm not sure how they do it in the Makefile; but what I'm guessing based on having seen it in other drivers in the past, they do a uname call and go based on that.  Unfortunately, on arm64 machines, the arch returned by uname -m is aarch64, however in the kernel sources, the arch is called arm64.

I grabbed the sources to take a look, and the problem line is in the Makefile - to fix it, you can change the line 
```
SUBARCH := $(shell uname -m | sed -e s/i.86/i386/ )
```
to
```
SUBARCH := $(shell uname -m | sed -e s/i.86/i386/ | sed -e s/aarch64/arm64/ )
```

This should probably go upstream to https://github.com/aircrack-ng/rtl8812au as an issue (and I'm sure if you added in a pull request making the change they would be happy to merge it!)

kimocoder, 2018-05-13:
Haven't seen this until now. Pushing the fix to the repo shortly. Thanks alot

steev, 2018-06-20:
Going to close this issue as the fix was pushed to that repo.  If you have issues please re-open!

closed 2018-06-20 by steev

muddy-13, 2018-06-26:
My 8812au still won't enter monitor mode.
iw phy0 info shows it capable of monitor mode....though lshw shows driverversion 3.14.79 firmware = N/A
I have installed bc as Steev sugested
and the current makefile is as Steev suggested from the current download.
SUBARCH := $(shell uname -m | sed -e s/i.86/i386/ | sed -e s/aarch64/arm64/ )

Make throws a lot of errors...... so something is still not right... another ARM64 reference somewhere? Or?
-----------------------------make output--------------------------------------------
oot@kali-arm64:/usr/src/realtek-rtl88xxau-5.1.5~20180513# make
make -j 4 ARCH=arm64 CROSS_COMPILE= -C /lib/modules/3.14.79/build M=/usr/src/realtek-rtl88xxau-5.1.5~20180513  modules
make[1]: Entering directory '/usr/src/kernel'

  WARNING: Symbol version dump /usr/src/kernel/Module.symvers
           is missing; modules will have no dependencies and modversions.

  CC [M]  /usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_cmd.o
  CC [M]  /usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_security.o
  CC [M]  /usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_debug.o
  CC [M]  /usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_io.o
/bin/sh: 1: /usr/src/kernel/scripts/recordmcount: Exec format error
scripts/Makefile.build:308: recipe for target '/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_io.o' failed
make[2]: *** [/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_io.o] Error 2
make[2]: *** Waiting for unfinished jobs....
/bin/sh: 1: /usr/src/kernel/scripts/recordmcount: Exec format error
scripts/Makefile.build:308: recipe for target '/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_security.o' failed
make[2]: *** [/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_security.o] Error 2
/bin/sh: 1: /usr/src/kernel/scripts/recordmcount: Exec format error
scripts/Makefile.build:308: recipe for target '/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_cmd.o' failed
make[2]: *** [/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_cmd.o] Error 2
/bin/sh: 1: /usr/src/kernel/scripts/recordmcount: Exec format error
scripts/Makefile.build:308: recipe for target '/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_debug.o' failed
make[2]: *** [/usr/src/realtek-rtl88xxau-5.1.5~20180513/core/rtw_debug.o] Error 2
Makefile:1278: recipe for target '_module_/usr/src/realtek-rtl88xxau-5.1.5~20180513' failed
make[1]: *** [_module_/usr/src/realtek-rtl88xxau-5.1.5~20180513] Error 2
make[1]: Leaving directory '/usr/src/kernel'
Makefile:1911: recipe for target 'modules' failed
make: *** [modules] Error 2
root@kali-arm64:/usr/src/realtek-rtl88xxau-5.1.5~20180513# 

Any ideas?

threeway, 2018-06-26:
Ah, yes, you’re running into the issue due to the cross compiling initially. 

You’ll want to go into /usr/src/kernel and run ‘make mrproper’

Then ‘cp /usr/src/odroid-c2.config /usr/src/kernel’

Then run ‘make modules_prepare’ in /usr/src/kernel

After that it should build. Essentially what happens is that the helper apps that are built in the kernel sources are compiled for amd64 because the kernel was cross compiled.

muddy-13, 2018-06-26:
Thanks... i will check that out in the morning.....

muddy-13, 2018-06-26:
It crashed missing a config file...
------------------------------------------------------------------------------------
root@kali-arm64:/usr/src/kernel# make mrproper
  CLEAN   .
  CLEAN   .tmp_versions
  CLEAN   scripts/basic
  CLEAN   scripts/dtc
  CLEAN   scripts/kconfig
  CLEAN   scripts/mod
  CLEAN   scripts/selinux/genheaders
  CLEAN   scripts/selinux/mdp
  CLEAN   scripts
  CLEAN   include/config include/generated arch/arm64/include/generated
  CLEAN   .config include/generated/uapi/linux/version.h
root@kali-arm64:/usr/src/kernel# cd ..
root@kali-arm64:/usr/src# cp odroid-c2.config /usr/src/kernel
root@kali-arm64:/usr/src# cd kernel
root@kali-arm64:/usr/src/kernel# make modules_prepare
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  SHIPPED scripts/kconfig/zconf.tab.c
  SHIPPED scripts/kconfig/zconf.lex.c
  SHIPPED scripts/kconfig/zconf.hash.c
  HOSTCC  scripts/kconfig/zconf.tab.o
  HOSTLD  scripts/kconfig/conf
scripts/kconfig/conf --silentoldconfig Kconfig
***
*** Configuration file ".config" not found!
***
*** Please run some configurator (e.g. "make oldconfig" or
*** "make menuconfig" or "make xconfig").
***
/usr/src/kernel/scripts/kconfig/Makefile:36: recipe for target 'silentoldconfig' failed
make[2]: *** [silentoldconfig] Error 1
/usr/src/kernel/Makefile:514: recipe for target 'silentoldconfig' failed
make[1]: *** [silentoldconfig] Error 2
make: *** No rule to make target 'modules_prepare'.  Stop.
root@kali-arm64:/usr/src/kernel# 
-----------------------------------------------------------------------------------------
I looked at the kconfig files and didn't see anything revealing....

steev, 2018-06-26:
You didn't see my edit, which likely wasn't sent to your email.

the correct step to get the config is `cp /usr/src/odroid-c2.config /usr/src/kernel/.config`

muddy-13, 2018-06-26:
I hadnt seen that .config !     So I ran that command.  I got a warning...1232.2:...syscall renameat2 not implemented [-Wcpp]
But otherwise the script completed. Still unable to get the 8812au into monitor mode.  Tried with NetworkManager stopped and started.  Did the check kill.... still drops the wlan0mon0 as soon as it creates it. Thanks Steev

muddy-13, 2018-06-26:
Interesting: iw dev wlan0 interface add mon0 type monitor.    Runs. Shows in airmon-ng with driver as????? And chipset non-mac80211 device. Airodump-ng runs without complaint.. but of course... sees no traffic.

kimocoder, 2018-06-26:
rtl8812au support monitor mode just fine. I know all about it ;)
It's just that airmon-ng isn't supported as the driver does not support virtual Interface.

Anyway, started to work on the newer v5.2.20 driver, which will replace the old v5.1.5 in some weeks.

Take a look [here](https://github.com/aircrack-ng/rtl8812au/tree/v5.2.20) and file issue reports to us og you got issues with rtl8812au.

Besides that, the latest build of v5.1.5 hasn't been pushed to Kali, and that includes a fix for ARM systems + Odroid build fix which may be seen [here](https://github.com/aircrack-ng/rtl8812au/commit/d628a7441c6066a2e302104b7f4f99bcd81a111f)

So you may clone the repo and build it or just pull that specific patch with "wget https://github.com/aircrack-ng/rtl8812au/commit/d628a7441c6066a2e302104b7f4f99bcd81a111f.patch"

steev, 2018-06-26:
@kimocoder just for the record, we do have 0513 in kali already.  The latest is 0608 which we don't have (yet)

kimocoder, 2018-06-26:
Aaaah yeah did see ut now.. my mistake there, even though I really should be the one to know it. 
Then I rest my case on this one.

jsl303, 2018-10-05:
I got further than before using the suggestions here, but I still can't get this to work.
I tried the following.
make mrproper
cp /usr/src/odroid-c2.config /usr/src/kernel/.config
make prepare
make modules_prepare  
Then I cloned the rtl8812au repo, and run "sudo ./dkms-install.sh".
However, I got "Error! Bad return status for module build on kernel: 3.14.79 (aarch64)".
Here's the log.
https://paste.ee/d/7hk1w
I also created an issue at rtl8812au
https://github.com/aircrack-ng/rtl8812au/issues/210
Thank you for your help!

steev, 2018-10-05:
Yeah, I'm not sure about that - that's not on the kali side of things - I'm not 100% but I'm pretty sure -mgeneral-regs-only isn't the correct flag to be passing, it should be passing -mfpu=none if they really insist on no floating point support (aarch64 requires neon support)