offensive-security/kali-arm-build-scripts #118: friendlyarm released a 4.4 kernel for nanopi3

daiaji, 2018-01-07:
it's here
http://wiki.friendlyarm.com/wiki/index.php/NanoPC-T3#Compile_Linux_kernel_4.4.y

Hope this will help you to update nanopi3.sh

steev, 2018-01-25:
Hi @daiaji, I noticed that, and as much as I want to move to it, at this time, I can't - video output doesn't work properly here.  The nanopi2 also seems to have issues working out the correct dtb file.  I've been keeping an eye on things, but for 2018.1, we're going to still be using the older kernel.

Once I sort out the video issues, 4.4 will be used (hopefully for 2018.2)

daiaji, 2018-01-27:
Hi @steev In fact there is an unofficial support for the 4.14 kernel his HDMI may be working properly
 https://github.com/rafaello7/linux-nanopi-m3-v4.14/
[image]
But it seems that the nanopi3 kernel has the problem that mono can not work
This is usually when I use https://github.com/Pryaxis/TShock
I am on the A9 A53 Android mobile phone mono is working properly
This made me almost give up using nanopi😪

rench, 2018-03-07:
@steev Could you please add NanoPi Neo Plus2 support?

steev, 2018-03-07:
@rench this is unrelated to your issue, please don't ask in unrelated issues.  as for adding support, i do not have the hardware but I can try to throw a script together that might work if you run it.

steev, 2018-10-27:
Hi @daiaji I'm going to stick with the 4.4 kernel for now as that's what the vendor suggests - you should be able to modify the script to build the 4.14 kernel.

closed 2018-10-27 by steev