offensive-security/kali-arm-build-scripts #119: Enabling SSH For Headless Setup?

jsl303, 2018-01-13:
Can we have an option to enable ssh and root login when running Kali headless?
For example in Raspbian, you can enable SSH by placing a file named “ssh” (without any extension) onto the boot partition of the SD card.
Can something like this be implemented?

steev, 2018-01-18:
ssh is already enabled on the arm images, which images are you having this issue on?

it may be possible to do it that way, but, i'd have to look into how they're doing so.

jsl303, 2018-01-19:
I'm having a problem with kali-2017.3-rpi3-nexmon.img.xz.

Re4son, 2018-02-12:
@steev: don't bother with the Raspbian way. They are running a service that looks in /boot for a file and enables the ssh service if present. I'm doing the same in my pi-tail. Very messy and you don't need it as the service is running already.

The issue with ssh are missing host keys.
We have to run "dpkg-reconfigure openssh-server" to generate them. Until two or three releases ago the image shipped with keys but not anymore.

steev, 2018-02-12:
Thanks for pointing it out @Re4son will pull now

steev, 2018-03-07:
Closing the issue as this is now fixed and released.

closed 2018-03-07 by steev