offensive-security/kali-arm-build-scripts #122: Would your please create a NanoPi NEO Plus2 image?

rench, 2018-02-26:
http://wiki.friendlyarm.com/wiki/index.php/NanoPi_NEO_Plus2

http://wiki.friendlyarm.com/wiki/index.php/Mainline_U-boot_and_Linux

steev, 2018-03-08:
As mentioned in the other thread that you commented on - I do not have the hardware, so I have no way to test if an image works.  What I can do, is attempt to create a script that would build the image, and push it to the git repository so that you'd be able to run it on a kali amd64 machine locally to build the image.

rench, 2018-03-08:
@steev 
Thank you very much for your reply, according to what you said, when you write the script, give me the script git address, I downloaded and then try to compile, I will give you the test results.

steev, 2018-10-27:
This was actually pushed out with 2018.3 and will be in 2018.4 as well - I'm looking into making an eflasher image as well, but so far have been unsuccessful at having something automated.

closed 2018-10-27 by steev