offensive-security/kali-arm-build-scripts #124: Raspberry 3 mod B

satoribeats, 2018-03-13:
Issues getting it to work.  Try'd all three versions on the website.  2.8 tft touch screen .  Thank you!

steev, 2018-06-20:
The only image that support the rpi3 at the moment is the rpi3-nexmon script.  You'll need to manually set up the 2.8 tft touch screen on it.

hoenheim93, 2018-07-05:
and for the model raspberry pi 3 B+ ?

threeway, 2018-07-05:
The 3b+ is supported by the rpi3 image

hoenheim93, 2018-07-05:
ok thanks , i flash sd  32 go now , after install kali ,how i can install lcd 3.5 (kuman)

hoenheim93, 2018-07-05:
1rst log , oblige use keyboard with usb ? not bluethooth ?

threeway, 2018-07-05:
I would suggest it since typically to pair a Bluetooth keyboard, you need do some setup. 

Please note, this is not a support forum for how to use the devices, you should be using the forums or IRC for that. 

This is more of a bug report/feature request for the scripts used to build the images

hoenheim93, 2018-07-05:
ok no worries you was very helpful, +1 and thank you again

hoenheim93, 2018-07-06:
?

steev, 2018-07-06:
Not private message.  Post to the forums - forums.kali.org or the #kali-linux irc channel on irc.freenode.net

hoenheim93, 2018-07-06:
Thank’s i go now

closed 2018-07-12 by steev