offensive-security/kali-arm-build-scripts #126: Beaglebone Black script for 2017.2 not booting  and BBB 2018.2 wifi problems

D4rkAngel74, 2018-05-01:
The image made by this script bbb.sh 2017.2 does not boot from the SD card.  I was able to access the image through chroot like explained here  [https://kali.training/topic/exercise-4-5-kali-linux-arm-chroot/](url) and even use apt-get update and install. but it will not boot. 

The pre-built image bbb 2018.2 will not connect SSH even after enabling SSH at boot, and when connecting the wifi, it wont connect even with the right PSK through NM or command line wpa_supplicant

D4rkAngel74, 2018-05-01:
The script bbb.sh has a keyring error when it runs, I got rid of all the apt -y --force -yes errors, but the image is still unable to boot
[image]

steev, 2018-06-05:
2018.2a is up and should fix this issue, please re-open if you still have the problem - the keyring issue is unrelated, and not an actual issue.

closed 2018-06-05 by steev