offensive-security/kali-arm-build-scripts #130: FriendlyArm NanoPC-T4

hanshaze, 2018-06-08:
**Hello.....
which script i have to choose for the NanoPC-T4 with lubuntu installed to build Kali Linux?**

```
pi@NanoPC:~$ uname -a
Linux NanoPC 4.4.126 #178 SMP Tue May 29 15:20:06 CST 2018 aarch64 GNU/Linux
```

steev, 2018-06-20:
None yet, this is on the roadmap for the future though!

hanshaze, 2018-06-25:
🙈🙉🙊

steev, 2018-06-25:
I don't have access to the hardware just yet.  It is ordered, just waiting on the shipment.

steev, 2018-10-27:
I have the hardware, I haven't had a chance to dig deeper just yet.  The rockchip kernel is very..... interesting, we will say.  Hoping to have something usable soon, but hard to say, I can't even get modules to insert properly.

CookieAndPizza, 2019-02-11:
Hi, has there been any progress on this? (: