offensive-security/kali-arm-build-scripts #132: Default Password toor for root Doesn't work on Fresh Image for Odroid C2?

jsl303, 2018-07-22:
I downloaded and flashed kali-linux-2018.2-odroidc2.img.
I booted up, and tried to log into ssh with username root and password toor.
It says access denied. What's the default password?

steev, 2018-07-22:
toor is the correct password.

If you have a second linux machine, can you mount the sdcard in it, and verify that the /etc/ssh/sshd_config file has "PermitRootLogin yes" in it and is uncommented?

steev, 2018-07-22:
Plug a monitor and keyboard in and check?

jsl303, 2018-07-23:
I'm visually impaired, , so I have to ssh from another computer with a screen reader.

steev, 2018-07-23:
Unfortunately, there isn't a quick fix then.  The image is missing that line in the config so you can't ssh in.

Is there a friend or someone you could walk through editing that file?  All they need to do is find the line that says 

#PermitRootLogin prohibit-password

and change it to

PermitRootLogin yes

jsl303, 2018-07-23:
Thanks for the explanation. Could we get this fixed in the next release 
though?

threeway, 2018-07-23:
It already is :) if you build an image with the script as it is now, you’ll get the fix. 

Just a warning that the first boot also resizes the file system and reboots to do so. It will also rebuild the ssh host keys.

threeway, 2018-07-23:
Also, if you build it yourself, you’ll want to run build-deps.sh first, to pull in everything that is needed. I haven’t updated the readme file to explain the changes just yet. Once I have all the images correctly building, I’ll be doing so.

jsl303, 2018-07-23:
Awesome, thank you!

steev, 2018-08-08:
I don't have an exact date unfortunately.  If you have a kali VM, you can clone the repository, run the build-deps script and then the odroid-c2 script if you cannot wait.

steev, 2018-10-29:
2018.4 is out, please give it a test, and re-open if you still have this issue!

closed 2018-10-29 by steev