offensive-security/kali-arm-build-scripts #134: Container  fails with error code 1

alex1005, 2018-08-15:
Hi,
I have  problem with while build the image for raspberry  pi 2. Building stops on the systemd-nspawn error massage "Container (name) failed with error code 1". The last massage before it is "I: Base system installed successfully". Please, help me. If you need more info, write me.

threeway, 2018-08-15:
Can you post the full output of the build script please?

alex1005, 2018-08-16:
Sorry for so long-time reply, I had some work to do.
Here my script's output:
```
The basedir thinks it is: /media/root/d/arm-stuff/kali-arm-build-scripts/rpi3-2.0
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id 44C6513A8E4FB3D30875F758ED444FF07D8D0BF6)
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Checking component main on http://http.kali.org/kali...
I: Retrieving adduser 3.117
I: Validating adduser 3.117
I: Retrieving apt 1.6.3
I: Validating apt 1.6.3
I: Retrieving apt-utils 1.6.3
I: Validating apt-utils 1.6.3
I: Retrieving base-files 1:2018.3.0
I: Validating base-files 1:2018.3.0
I: Retrieving base-passwd 3.5.45
I: Validating base-passwd 3.5.45
I: Retrieving bash 4.4.18-3.1
I: Validating bash 4.4.18-3.1
I: Retrieving bsdmainutils 11.1.2+b1
I: Validating bsdmainutils 11.1.2+b1
I: Retrieving bsdutils 1:2.32-0.1
I: Validating bsdutils 1:2.32-0.1
I: Retrieving coreutils 8.28-1
I: Validating coreutils 8.28-1
I: Retrieving cpio 2.12+dfsg-6
I: Validating cpio 2.12+dfsg-6
I: Retrieving cron 3.0pl1-130
I: Validating cron 3.0pl1-130
I: Retrieving dash 0.5.8-2.10
I: Validating dash 0.5.8-2.10
I: Retrieving debconf 1.5.69
I: Validating debconf 1.5.69
I: Retrieving debconf-i18n 1.5.69
I: Validating debconf-i18n 1.5.69
I: Retrieving debian-archive-keyring 2017.7
I: Validating debian-archive-keyring 2017.7
I: Retrieving debianutils 4.8.6
I: Validating debianutils 4.8.6
I: Retrieving diffutils 1:3.6-1
I: Validating diffutils 1:3.6-1
I: Retrieving dmidecode 3.1-1
I: Validating dmidecode 3.1-1
I: Retrieving dmsetup 2:1.02.145-4.1
I: Validating dmsetup 2:1.02.145-4.1
I: Retrieving dpkg 1.19.0.5kali1
I: Validating dpkg 1.19.0.5kali1
I: Retrieving e2fsprogs 1.44.3-1
I: Validating e2fsprogs 1.44.3-1
I: Retrieving fdisk 2.32-0.1
I: Validating fdisk 2.32-0.1
I: Retrieving findutils 4.6.0+git+20171230-2
I: Validating findutils 4.6.0+git+20171230-2
I: Retrieving gcc-8-base 8.1.0-12
I: Validating gcc-8-base 8.1.0-12
I: Retrieving gdbm-l10n 1.14.1-6
I: Validating gdbm-l10n 1.14.1-6
I: Retrieving gpgv 2.2.9-1
I: Validating gpgv 2.2.9-1
I: Retrieving grep 3.1-2
I: Validating grep 3.1-2
I: Retrieving gzip 1.6-5+b1
I: Validating gzip 1.6-5+b1
I: Retrieving hostname 3.20
I: Validating hostname 3.20
[... 612 lines cut ...]
I: Configuring cron...
I: Configuring libmnl0:armhf...
I: Configuring logrotate...
I: Configuring rsyslog...
I: Configuring libip6tc0:armhf...
I: Configuring netbase...
I: Configuring vim-tiny...
I: Configuring libnetfilter-conntrack3:armhf...
I: Configuring debconf-i18n...
I: Configuring libisc-export169:armhf...
I: Configuring libiptc0:armhf...
I: Configuring iproute2...
I: Configuring iptables...
I: Configuring libnewt0.52:armhf...
I: Configuring libdns-export1102...
I: Configuring ifupdown...
I: Configuring whiptail...
I: Configuring isc-dhcp-client...
I: Configuring tasksel...
I: Configuring tasksel-data...
I: Configuring libc-bin...
I: Configuring systemd...
I: Base system installed successfully.
Container ScIT2lfalNYQIlbt failed with error code 1.
```

closed 2018-08-16 by alex1005