offensive-security/kali-arm-build-scripts #136: Create a arm64 script for the rapi 3b and 3b+?

peterfarge, 2018-09-19:
Hi,

it would be nice to have a script for the raspberry 3 b or b+ based on 64 bit.

Thanks a lot

Peter

threeway, 2018-09-19:
Pull requests are accepted. 

Ive poked at it a few times but have never gotten it to boot properly. 

The main changes would be to add the bits to make sure the firmwares switches to 64bit mode, the kernel and then changing the architecture=“arm64”, for pointers to help you on your way :)

steev, 2018-10-13:
Good news, I've added it.  I tested it with a B+ - the one thing you will need to do, is edit the config.txt after imaging the sdcard and making sure the device_tree= line is uncommented for the board you have, there is no auto detection of the board right now, apparently.

I've commented which device_tree= line is which, and the 3B, 3B+ and CM3 modules are all supported.  It pulls in the 4.14.71 re4son kernel, and has nexmon support.

If you could give a build and test a whirl, that would be great!

closed 2018-10-13 by steev