offensive-security/kali-arm-build-scripts #138: `apt-get update` reports "Release file not valid yet."

kvesel, 2018-10-10:
I am using kali-linux-2018.3-rpi3-nexmon image from https://www.offensive-security.com/kali-linux-arm-images/

I am having Raspberry Pi 3b+ with booting working.

```
root@kali:~# apt-get -y update
Get:1 http://kali.download/kali kali-rolling InRelease [30.5 kB]
Reading package lists... Done
E: Release file for http://http.kali.org/kali/dists/kali-rolling/InRelease is not valid yet (invalid for another 109d 18h 50min 51s). Updates for this repository will not be applied.
root@kali:~#
```
Is maybe image wrong or why error? Google searching says problems for Ubuntu and Debian sometimes like this, but solution is to remove repository. This is only repository I am thinking, so I cannot remove?

To note I do not need specific imaging for nexmon, any working kali armhf on rpi3b+ is okay.

kvesel, 2018-10-10:
Correction only error is when: sudo apt-get -y update
but not for: apt-get -y update

dogukanerel, 2019-03-11:
You have wrong timezone set. You can fix it or use this temporarily:
sudo apt-get -o Acquire::Check-Valid-Until=false update