offensive-security/kali-arm-build-scripts #140: Can't Install raspi3-firmware

jsl303, 2018-10-14:
I flashed kali-linux-2018.3-rpi3-nexmon.img on Raspberry Pi  B+.
Then I got the following error during apt dist-upgrade.
Setting up raspi3-firmware (1.20180619-1) ...
Error: missing /boot/firmware, did you forget to mount it?
dpkg: error processing package raspi3-firmware (--configure):
 installed raspi3-firmware package post-installation script subprocess returned error exit status 1
Errors were encountered while processing:
 raspi3-firmware

E: Sub-process /usr/bin/dpkg returned an error code (1)

There is firmware folder inside /boot, but it's empty.

steev, 2018-10-15:
Correct - this is actually covered in the forums already; and dealt with in the script (temporary workaround while trying to come up with a  better fix - it has to do with the way Debian does their partitioning which is different than upstream (and by extension us, despite our reliance on Debian.)

https://forums.kali.org/showthread.php?41853-Raspberry-Pi-3-B-Error-During-dist-upgrade is the forum post to cover it, but the basic gist reproduced here is to

```
mount /dev/mmcblk0p1 /boot
mkdir -p /boot/firmware
mount -t tmpfs /boot/firmware
```
And then do the dist-upgrade

To make the workaround more permanent add 

`tmpfs         /boot/firmware tmpfs rw 0 0`

to your /etc/fstab so it is always mounted.

In the future, something like this is better covered under either bugs.kali.org or in the forums, the issues here are more for dealing with the scripts themselves.

closed 2018-10-15 by steev

jsl303, 2018-10-16:
Thanks for the link and pointing out where to post.
Do bugs.kali.org deal with Raspberry specific problem as well?
For example, Pi 3 B+ builtin WiFi doesn't work with Kali Pi image for 
me, but it works with raspbian stretch.
Should I file the bug here or bugs.kali.org?
Thanks,

steev, 2018-10-16:
If you ran dist-upgrade, you probably ran into the issue that I left the firmware-brcm80211 package installed, and the upgrade wrote over the files.

The fix there is to 

For nexmon firmware:
```
apt purge firmware-brcm80211
cd /lib/firmware/brcm
cp brcmfmac43430-sdio.nexmon.bin brcmfmac43430-sdio.bin
cp brcmfmac43455-sdio.nexmon.bin brcmfmac43455-sdio.bin
```

Or if you don't want to use the nexmon firmware:
```
apt purge firmware-brcm80211
cd /lib/firmware/brcm
cp brcmfmac43455-sdio.rpi.bin brcmfmac43455-sdio.bin
cp brcmfmac43430-sdio.rpi.bin brcmfmac43430-sdio.bin
```

IIRC, there was already a bug filed, but there was also a forum post for that.

The next release of the images (2018.4) will have it fixed already.