offensive-security/kali-arm-build-scripts #141: Fixed errors of pi-bluetooth for Pi Zero W

mame82, 2018-10-17:
In my tests, the kali builds for Pi0w end up with a not working hci device (the bdaddr of the adapter shows as "AA:AA:AA:AA:AA:AA" with `hcitool dev`).

This is caused by not loading the firmware correctly. A detailed description could be found here:
https://github.com/Re4son/re4son-kernel-builder/issues/7

Re4son kindly provided an updated deb package here:
https://raw.githubusercontent.com/Re4son/re4son-kernel-builder/build-4.14.n/repo/pi-bluetooth+re4son_2.2_all.deb

This has to be tested, as I'm working with a custom build-script, but currently I'm using the exact same commands as in this repo for the bluetooth part.

Additionally the following files are already part of the aforementioned deb package:
```
/etc/udev/rules.d/99-com.rules
/lib/systemd/system/hciuart.service
/usr/bin/btuart
```

So it is not necessary to add them in in the rpi0-* build-scripts, like done here:
https://github.com/offensive-security/kali-arm-build-scripts/blob/master/rpi0w-nexmon.sh#L203

mame82, 2018-10-17:
Additional note:
Although the link to the debian package of re4son implies a strict kernel dependency on kernel 4.14.x, the package more or less depends on the Bluez version in use (userland part).
Since Bluez 5.37 the `btattach` tool is provided with Bluez (see https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/ChangeLog#n165). In contrast to `hciattach` this tool is working on the Kali builds for RPI0w, to bring up the hciuart and preload the proper firmware (for BCM43430A1).

steev, 2018-10-17:
Good catches, and thanks for the heads up on the package update.  I've updated to the package and removed the redundant files (they should have been overwritten by the package anyway, but might as well remove them since they aren't needed)

I have only tested that it builds so far, will be testing it soon.

steev, 2018-10-27:
Tested here and everything seems good.  Once 2018.4 is out, let me know if you still have issues.

closed 2018-10-27 by steev