offensive-security/kali-arm-build-scripts #144: GPIO not working on Kali Linux RaspberryPi 3 64bit 2018.4a

dingyifei, 2018-11-21:
The RPi.GPIO is not working on Kali Linux RaspberryPi 3 64bit 2018.4a 
I install the image download from https://www.offensive-security.com/kali-linux-arm-images/ and when I try to import RPi.GPIO on python 3.7.1 and python3.6, I receive the following error:
```
root@kali:~# python3
Python 3.6.7 (default, Oct 21 2018, 08:08:16) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import RPi.GPIO
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/lib/python3/dist-packages/RPi/GPIO/__init__.py", line 23, in <module>
    from RPi._GPIO import *
RuntimeError: This module can only be run on a Raspberry Pi!
>>> 
```
Already tried reinstall RPi.GPIO
I tried install wiringpi and when I use `gpio readall` I get the following error: 
```
root@kali:~# gpio readall
Oops: Unable to determine board revision from /proc/cpuinfo
 -> No "Hardware" line
 ->  You'd best google the error to find out why.
```
Here are the `cat /proc/cpuinfo` 
```
root@kali:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

steev, 2018-11-21:
Did you compile it yourself? Afaik, the packages have binaries that aren’t 64bit

dingyifei, 2018-11-21:
I followed the procedure in `http://wiringpi.com/download-and-install/` compiled WiringPi by clone and run the build script. I didn't compile the rpi.gpio-common... isn't rpi.gpio a python program?

dingyifei, 2018-11-21:
The rpi.gpio-common, python3-rpi.gpio, and python-rpi.gpio are all arm64 package base on apt search rpi.gpio result

digitallyelite, 2018-11-21:
ah you're right - it looks like the 64bit package doesn't actually pass the check for RPI for some reason.

Can you give me the output of /proc/cpuinfo on yours?

Keep in mind, the RPi Foundation does not support running the RPi in 64bit mode at all, so we're gonna have to do a little leg work.  This will likely have to be passed on to a debian bug as we (in kali) don't modify the package from Debian at all, but lets do the troubleshooting before we do so.

dingyifei, 2018-11-21:
The raspberry pi I'm using is `Raspberry Pi 3 B+` (the one with higher clock speed cpu and metal shielding on the processor)
I tested the `kali-linux-2018.4-rpi3-nexmon.img` and it is working fine
Here are the `cat /proc/cpuinfo` output:
```
root@kali:~# cat /proc/cpuinfo
processor       : 0
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 1
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 2
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4

processor       : 3
BogoMIPS        : 38.40
Features        : fp asimd evtstrm crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant     : 0x0
CPU part        : 0xd03
CPU revision    : 4
```

digitallyelite, 2018-11-21:
Right - Basically what we're looking at is https://sourceforge.net/p/raspberry-gpio-python/code/ci/default/tree/source/c_gpio.c - and seeing if we can massage it to see that we're an RPi - since I'm guessing we lack the Hardware: line in 64bit that is in 32bit.

dingyifei, 2018-11-21:
I get some completely different cpuinfo from the 32 bit kali `cat /proc/cpuinfo `:
```processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 38.40
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a020d3
Serial		: 000000004be218c8

```

digitallyelite, 2018-11-21:
Right, in 32bit, it has the Hardware/Revision/Serial stanza.

And the rpi.gpio package looks for that to tell if it's on an RPi or not.

As far as I know, there's no way to have the Hardware/Revision/Serial stanza in arm64, at least, not without someone patching the kernel to add it.

dingyifei, 2018-11-21:
Im looking at the source code form `https://pypi.org/project/RPi.GPIO/` maybe just set the value in it to BCM2835?

digitallyelite, 2018-11-21:
you could hack it locally to always return true, yes, but unfortunately, that won't work for a debian package, since it can be installed on any (arm64) system.

dingyifei, 2018-11-21:
I get it to work LOL
The following are the modified \RPi.GPIO-0.6.5\source\cpuinfo.c
```
/*
Copyright (c) 2012-2016 Ben Croston

Permission is hereby granted, free of charge, to any person obtaining a copy of
this software and associated documentation files (the "Software"), to deal in
the Software without restriction, including without limitation the rights to
use, copy, modify, merge, publish, distribute, sublicense, and/or sell copies
of the Software, and to permit persons to whom the Software is furnished to do
so, subject to the following conditions:

The above copyright notice and this permission notice shall be included in all
copies or substantial portions of the Software.

THE SOFTWARE IS PROVIDED "AS IS", WITHOUT WARRANTY OF ANY KIND, EXPRESS OR
IMPLIED, INCLUDING BUT NOT LIMITED TO THE WARRANTIES OF MERCHANTABILITY,
FITNESS FOR A PARTICULAR PURPOSE AND NONINFRINGEMENT. IN NO EVENT SHALL THE
AUTHORS OR COPYRIGHT HOLDERS BE LIABLE FOR ANY CLAIM, DAMAGES OR OTHER
LIABILITY, WHETHER IN AN ACTION OF CONTRACT, TORT OR OTHERWISE, ARISING FROM,
OUT OF OR IN CONNECTION WITH THE SOFTWARE OR THE USE OR OTHER DEALINGS IN THE
SOFTWARE.
*/

#include <stdio.h>
#include <stdlib.h>
#include <string.h>
#include "cpuinfo.h"

int get_rpi_info(rpi_info *info)
{
    FILE *fp;
    char buffer[1024];
    char hardware[1024];
    char revision[1024];
    char *rev;
    int found = 0;
    int len;

    char fake_revision[1024] = "a020d3";
    char fake_hardware[1024] = "BCM2835";
    found = 1;
    
    if ((fp = fopen("/proc/cpuinfo", "r")) == NULL)
        return -1;
    while (!feof(fp) && fgets(buffer, sizeof(buffer), fp))
    {
        sscanf(buffer, "Hardware	: %s", fake_hardware);
        if (strcmp(fake_hardware, "BCM2708") == 0 ||
            strcmp(fake_hardware, "BCM2709") == 0 ||
            strcmp(fake_hardware, "BCM2835") == 0 ||
            strcmp(fake_hardware, "BCM2836") == 0 ||
            strcmp(fake_hardware, "BCM2837") == 0)
        {
            found = 1;
        }
        sscanf(buffer, "Revision	: %s", fake_revision);
    }

    fclose(fp);

    if (!found)
        return -1;

    if ((len = strlen(fake_revision)) == 0)
        return -1;

    if (len >= 6 && strtol((char[]){fake_revision[len - 6], 0}, NULL, 16) & 8)
    {
[... 460 lines cut ...]
        return self._bits_per_word

    def _set_bits_per_word(self, value):
        if value < 1:
            raise ValueError('bits_per_word must be positive')
        self._bits_per_word = int(value)

    def _get_select_high(self):
        return self.active_high

    def _set_select_high(self, value):
        with self._bus.lock:
            self.active_high = value
            self.off()

class LocalPiHardwareSPIShared(SharedMixin, LocalPiHardwareSPI):
    @classmethod
    def _shared_key(cls, factory, port, device):
        return (port, device)

class LocalPiSoftwareSPIShared(SharedMixin, LocalPiSoftwareSPI):
    @classmethod
    def _shared_key(cls, factory, clock_pin, mosi_pin, miso_pin, select_pin):
        return (select_pin,)
```

closed 2018-11-21 by dingyifei