offensive-security/kali-arm-build-scripts #145: Trying to build RPI3-Nexmon and the build fails in trying to access raspberrypi-kernel

open-analysis, 2018-12-06:
I’ve been trying build a rpi3-nexmon script to build, but it fails in the third_stage. The error is: “dpkg: error: can not access archive ‘root/raspberrypi-kernel_20181124-191322_armhf.deb’ : No such file or directory” However, the problem is that the file does exist but it can’t seeem to do anything. What needs to be done fix this and finishing building the dependency?

steev, 2018-12-06:
That seems odd, the script downloads that file and puts it where it needs to be - are you maybe running out of space on the drive your running the script from?  That’s the only thing I could think may be doing it.

open-analysis, 2018-12-06:
> That seems odd, the script downloads that file and puts it where it needs to be - are you maybe running out of space on the drive your running the script from? That’s the only thing I could think may be doing it.

The computer this script is running on has a 10GB of unused disk space so there should be plenty of room spare. Is there anything else that might be interfering the script completing?

steev, 2018-12-06:
Perhaps a flakey internet connection?

If you could run the build again, and redirect the output to a file and paste it, I could see - I cannot reproduce the issue at all here, either running it manually nor through my automated builds (in fact, the automated build just completed a run 1 minute ago.)

Simply doing something like `./rpi3-nexmon.sh test 2>&1 | tee failing-build.log` and then paste the outputs into a comment here so I can see what your system is doing?  If you've modified the file at all, it would be nice to see the changes as well.

open-analysis, 2018-12-07:
The changes I made are from the Kali official walk-through: https://kali.training/topic/exercise-4-4-custom-kali-linux-arm-install/
The changes I made are:
`#desktop="fonts-croscore fonts-crosextra-caladea fonts-crosextra-carlito gnome-theme-kali gtk3-engines-xfce kali-desktop-xfce kali-root-login lightdm network-manager network-manager-gnome xfce4 xserver-xorg-video-fbdev xserver-xorg-input-evdev xserver-xorg-input-synaptics"
#tools="aircrack-ng ethtool hydra john libnfc-bin mfoc nmap passing-the-hash sqlmap usbutils winexe wireshark net-tools"
tools="aircrack-ng nmap hostapd"
#services="apache2 openssh-server"
services="openssh-server"
#extras="iceweasel xfce4-terminal wpasupplicant python-smbus i2c-tools python-requests python-configobj python-pip"

#packages="${arm} ${base} ${desktop} ${tools} ${services} ${extras}"
packages="${arm} ${base} ${tools} ${services}"`

Then under the WiFi section:
`auto eth0
        iface eth0 inet static
        address 192.168.1.12
        netmask 255.255.255.0
        gateway 192.168.1.1
EOF`

The Log:
`The basedir thinks it is: /root/arm-stuff/kali-arm-build-scripts/rpi3-nexmon-test
I: Retrieving InRelease 
I: Checking Release signature
I: Valid Release signature (key id 44C6513A8E4FB3D30875F758ED444FF07D8D0BF6)
I: Retrieving Packages 
I: Validating Packages 
I: Resolving dependencies of required packages...
I: Resolving dependencies of base packages...
I: Checking component main on http://http.kali.org/kali...
I: Retrieving adduser 3.118
I: Validating adduser 3.118
I: Retrieving apt 1.8.0~alpha2
I: Validating apt 1.8.0~alpha2
I: Retrieving apt-utils 1.8.0~alpha2
I: Validating apt-utils 1.8.0~alpha2
I: Retrieving base-files 1:2019.1.1
I: Validating base-files 1:2019.1.1
I: Retrieving base-passwd 3.5.45
I: Validating base-passwd 3.5.45
I: Retrieving bash 4.4.18-3.1
I: Validating bash 4.4.18-3.1
I: Retrieving bsdmainutils 11.1.2+b1
I: Validating bsdmainutils 11.1.2+b1
I: Retrieving bsdutils 1:2.32.1-0.1
I: Validating bsdutils 1:2.32.1-0.1
I: Retrieving coreutils 8.30-1
I: Validating coreutils 8.30-1
I: Retrieving cpio 2.12+dfsg-6
I: Validating cpio 2.12+dfsg-6
I: Retrieving cron 3.0pl1-130
I: Validating cron 3.0pl1-130
I: Retrieving dash 0.5.10.2-1
I: Validating dash 0.5.10.2-1
I: Retrieving debconf 1.5.69
I: Validating debconf 1.5.69
I: Retrieving debconf-i18n 1.5.69
I: Validating debconf-i18n 1.5.69
I: Retrieving debian-archive-keyring 2018.1
I: Validating debian-archive-keyring 2018.1
I: Retrieving debianutils 4.8.6
I: Validating debianutils 4.8.6
I: Retrieving diffutils 1:3.6-1
I: Validating diffutils 1:3.6-1
I: Retrieving dmidecode 3.2-1
I: Validating dmidecode 3.2-1
I: Retrieving dmsetup 2:1.02.145-4.1
I: Validating dmsetup 2:1.02.145-4.1
I: Retrieving dpkg 1.19.2kali1
I: Validating dpkg 1.19.2kali1
[... 2837 lines cut ...]
Adding 'diversion of /boot/overlays/uart0.dtbo to /usr/share/rpikernelhack/overlays/uart0.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/uart1.dtbo to /usr/share/rpikernelhack/overlays/uart1.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/upstream-aux-interrupt.dtbo to /usr/share/rpikernelhack/overlays/upstream-aux-interrupt.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/upstream.dtbo to /usr/share/rpikernelhack/overlays/upstream.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/vc4-fkms-v3d.dtbo to /usr/share/rpikernelhack/overlays/vc4-fkms-v3d.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/vc4-kms-kippah-7inch.dtbo to /usr/share/rpikernelhack/overlays/vc4-kms-kippah-7inch.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/vc4-kms-v3d.dtbo to /usr/share/rpikernelhack/overlays/vc4-kms-v3d.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/vga666.dtbo to /usr/share/rpikernelhack/overlays/vga666.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/w1-gpio-pullup.dtbo to /usr/share/rpikernelhack/overlays/w1-gpio-pullup.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/w1-gpio.dtbo to /usr/share/rpikernelhack/overlays/w1-gpio.dtbo by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/waveshare32b-overlay.dtb to /usr/share/rpikernelhack/overlays/waveshare32b-overlay.dtb by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/waveshare32b-overlay.dts to /usr/share/rpikernelhack/overlays/waveshare32b-overlay.dts by rpikernelhack'
dpkg-divert: warning: please specify --no-rename explicitly, the default will change to --rename in 1.20.x
Adding 'diversion of /boot/overlays/waveshare32b.dtbo to /usr/share/rpikernelhack/overlays/waveshare32b.dtbo by rpikernelha

threeway, 2018-12-07:
Based on that log file, there was an isssue downloading nmap, not the kernel, or anything kernel related. 

Either the mirror was in the middle of updating, or the mirror you got has a few out of date files - can you try hardcoding a different mirror? The “mirror=http.kali.org” line is the one to change. 

Also please be sure you didn’t accidentally remove the beginning slash for the dpkg line for the kernel - you can quickly see your changes by doing “git diff” in the kali-arm-build-scripts checkout

open-analysis, 2018-12-08:
I switched the mirror and it still wasn't working, so I decided to remove nmap from the installation. Also the dpkg line wasn't changed since I didn't see anything related to a line like that when I typed "git diff".

The newer installation without nmap seemed to work better, however still ran into a problem with inability to access a file. Will update further on Monday. Thank you all for the help thus far.

open-analysis, 2018-12-09:
[failing-build.log](https://github.com/offensive-security/kali-arm-build-scripts/files/2661143/failing-build.log)
I've changed the mirror in the file, but nmap still seems to be causing trouble, even removing it from the tools didn't help. The final error at the end of the file says that file can't read sshd_config, I'm not sure if this is related to the fact that the previous issue.

open-analysis, 2018-12-09:
[failing-build.log](https://github.com/offensive-security/kali-arm-build-scripts/files/2661147/failing-build.log)
I've changed the mirror and removed the installation of tools just in case. Now however, it cannot find the firmware folder to get the stats off of. I've scanned through the rest of the log file and there weren't any other errors as far as I could see.

threeway, 2018-12-10:
Can you try the mirror “kali.mirror.globo.tech” ? I just checked and that’s the mirror that the build system gets as the round robin from http.kali.org

As to the second failed build - that’s because you disabled it installing raspi3-firmware - that package installs the rpi3 firmware and puts it in /boot/firmware and since we use a different layout (we have the first partition as /boot to mirror raspbian to be closer to what their scripts and all the instructions out there use) it’s failing to copy them. 

Interestingly enough, your failure of nmap, is a 403 forbidden, not a 404 not found. 

I should move the raspi3-firmware to a different line to be sure it’s installed since it is important and required, without it, an image won’t boot unless someone modified the script to clone the upstream firmware repo and move the files into /boot.

threeway, 2018-12-10:
Also to address the mention of sshd_config in your previous comment - because there’s an issue installing nmap, the packages fail to install and because openssh-server didn’t install (due to the nmap package failing), the /etc/ssh/sshd_config file doesn’t exist. 

I have the script failing due to that because previously, I didn’t have it fail, and since the packages aren’t installed, the build was succeeding but (obviously) not working properly

open-analysis, 2018-12-10:
[failing-build.log](https://github.com/offensive-security/kali-arm-build-scripts/files/2664906/failing-build.log)
I tried the mirror that you mentioned and was able to install nmap with it, and I made sure the rpi3 firmware was installed as well this time. I believe that the script ran properly this time as I ended up with an .img file, however, based on the [Kali Revealed](https://kali.training/topic/exercise-4-4-custom-kali-linux-arm-install/), there should be a zipped .img file. Which is correct? The .img.xz or just a plain .img, or is something else up?

steev, 2018-12-10:
It’s img.xz - the image (img file) is compressed with xz because it compresses a bit better than a zip, but it doesn’t roll off the tongue as well as “zipped” which is often used synonymously as a compressed file. Although I don’t see anything on the page about a zipped file. 

The only time you won’t have an img.xz file, is if you do it on 32bit, as 32bit will run out of memory compressing.  The big change from when that was written until now is that it writes the image into the directory you run the script from so you don’t have to look in the folder to see it.

open-analysis, 2018-12-11:
[image]
(^ the output of the script, clicking it will turn it right side up)
My apologies, to clarify: After the script stopped running, the files that were in the rpi3-nexmon-1.0 folder were the file system (kali-armhf), the image, nexmon folder, and the root folder. Nothing in the log file jumped out at me as saying something went wrong for it to just stop before compressing the image.

digitallyelite, 2018-12-19:
i'm not sure what would cause that, aside from it not fully running correctly - do you have the output of the build?  I run automated builds of all the rpi scripts and get a notification if something fails pretty quick, so the only thing i can think is another issue with downloading somewhere.  You can run the script with something like `./rpi3-nexmon.sh 1.0 2>&1| tee rpibuild.log` and it would produce a file with the output that I could look through.

open-analysis, 2018-12-19:
[build.log](https://github.com/offensive-security/kali-arm-build-scripts/files/2693219/build.log)
Here's the output from the build. I've only changed the mirror to what was suggested above and removed the desktop in the included packages and changed how the network interface will be setup, which I wouldn't think would change anything.

steev, 2018-12-19:
Can you try to add a `sleep 10` after the `sync` and before the `umount -l ${bootp}`  (line 459). I wonder if your cache isn't actually finished writing - it's not unmounting the boot partition yet even though it's unmounting the second partition.

steev, 2018-12-19:
Okay, I'm not entirely certain but this may be my fault in not noticing that the size of the install was increasing - Can you set the size (line 21) to 5000 instead of 4500 and see if that works?  I started running into a similar issue last night/today with the automated builds but it was throwing an error that it was running out of space not that it couldn't unmount.

I need to go through that walkthrough and see about getting modifications in to match the latest scripts, it's possible that we're still installing things even though you're not adding them (due to dependencies).

steev, 2018-12-19:
No, I'm wrong - your build script log output even shows you're only using 1.6GB of space.

What kind of hardware is being used here?  The script should be lazily unmounting, so the loop device *shouldn't* be in use still.

open-analysis, 2018-12-20:
I'm using a virtual machine that's running Kali which I made sure to give more than enough RAM and hard drive space. Otherwise, I have a i5 processor and 6GB of RAM.

steev, 2018-12-20:
What type of hard drive?  It seems like somehow its slow?  I'll admit, I
haven't tested the scripts extensively in a VM, but they ran successfully
last time I did them.  Is it virtualbox or vmware?

open-analysis, 2018-12-20:
I'm use virtualbox and my [hard drive](https://www.newegg.com/Product/Product.aspx?Item=9SIA4M58G94390&Description=hgst%20hdn724040ale640&cm_re=hgst_hdn724040ale640-_-9SIA4M58G94390-_-Product) has 7,000 RPM and 6Gb/s.

steev, 2019-01-02:
I can't seem to replicate this here, are you still having issues with it?

open-analysis, 2019-01-04:
Yes I am unfortunately. I've tried increasing the sleep time to no avail and trying to run the scrip on another Linux machine I had lying around but that died part way through, so I've been out of luck.