offensive-security/kali-arm-build-scripts #151: rpi0: g_serial and /dev/ttyGS0

rmoriz, 2019-01-16:
Raspberry Pi Zero provides nice OTG modes to use ethernet or serial over USB/OTG. A good documentation is available at https://gist.github.com/gbaman/50b6cca61dd1c3f88f41

This is very handy in headless usage/install, especially when wifi is not working/available.

However when enabled, the device is `/dev/ttyGS0` and is not whitelisted by default in `/etc/securetty`. So `root` can't log in. Given that kali has no non-root user by default (like `pi` in stock Raspbian)  this prevents login:

```
Jun 22 11:15:04 kali login[306]: pam_securetty(login:auth): access denied: tty '/dev/ttyGS0' is not secure !
Jun 22 11:15:08 kali login[306]: FAILED LOGIN (1) on '/dev/ttyGS0' FOR 'root', Authentication failure
```

steev, 2019-01-16:
Interesting find.  I'll see about adding this in for the next release!

steev, 2019-01-21:
I've pushed this to the rpi0w script so it will be in the 2019.1 release.

closed 2019-01-21 by steev